balena-io-library/base-images #7: Python 3 base image

techniq, 2015-08-16:
I was looking to submit a pull request to generate Python 3 base images similar to how Node is setup to generate all the different versions, but it looks a little more involved than I was expecting.  Any chance of getting this added or could someone help point me in the various locations to do so?

It looks like the following places would need updated:
- Update the scripts and templates under https://github.com/resin-io-library/base-images/tree/master/python
- Update the official library repos - https://github.com/resin-io-library/official-images/tree/master/library

I'm still trying to understand how they are built (bashbrew) and how the git repos relate to the images built

nghiant2710, 2015-08-16:
Hi @techniq,

Please submit your PR which should include templates for python3 images under https://github.com/resin-io-library/base-images/tree/master/python and this script https://github.com/resin-io-library/base-images/blob/master/python/generate-dockerfile.sh. Then we will take care the rest.

About how we build the image, you can read from this build script to know more: https://github.com/resin-io-library/official-images/blob/master/automation/jenkins_build.sh. Bashbrew will build the Dockerfile from the links inside library file under https://github.com/resin-io-library/official-images/tree/master/library.

nghiant2710, 2015-11-18:
Python3 images are available!

closed 2015-11-18 by nghiant2710

techniq, 2015-11-18:
Awesome, thanks. I had started work on this a few months ago and never got around to finishing it. I appreciate the work.