balena-io-library/base-images #118: invalid hostename, denied permission, /tmp/console not found on RPi3

hinzundcode, 2016-04-27:
I get this errors/warnings if I try to run resin/raspberrypi3-node:latest on my raspberry pi 3 under arch linux:

```
$ docker run --rm -it resin/raspberrypi3-node node
Unable to find image 'resin/raspberrypi3-node:latest' locally
latest: Pulling from resin/raspberrypi3-node
[...]
Digest: sha256:f4a53345937ad416ff8adfad8e40476dc59a40fc751ba4133230cbed5aecbdfd
Status: Downloaded newer image for resin/raspberrypi3-node:latest
hostname: the specified hostname is invalid
mount: permission denied
mount: permission denied
mount: permission denied
mount: permission denied
mount: mount point /tmp/console is not a directory
umount: /dev: must be superuser to unmount
mount: permission denied
> 
```

nghiant2710, 2016-04-27:
@hinzundcode, in our base images, we have an entry script which is called on start (ENTRYPOINT in dockerfile), it requires container to be run with privileged mode.
Please add `--privileged` in your `docker run` command and these log will go away

hinzundcode, 2016-04-27:
Thanks for your quick reply. I still get the hostname error. Any ideas?
```
$ docker run --rm --privileged -it resin/raspberrypi3-node node
hostname: the specified hostname is invalid
> 
```

nghiant2710, 2016-04-28:
@hinzundcode this is how we set hostname
```
HOSTNAME=$(cat /etc/hostname)-${RESIN_DEVICE_UUID:0:6}
echo $HOSTNAME > /etc/hostname
echo "127.0.1.1 $HOSTNAME" >> /etc/hosts
hostname $HOSTNAME
```
Since you're not running in resin device, so `RESIN_DEVICE_UUID` is not specified. So the `HOSTNAME` in this case will end with `-` which is invalid ending character for host name.

You can ignore this hostname error because your container is still running fine Or you can replace our entry script.

nghiant2710, 2016-04-28:
@hinzundcode we will update our script and detect if `RESIN_DEVICE_UUID` is not there. So you won't have this annoying issue anymore

hinzundcode, 2016-04-28:
Thanks a lot!

closed 2016-04-28 by hinzundcode