balena-io-library/base-images #189: udevd use a lot

Sid3y1, 2016-10-05:
I've a situation with udevd witch take all the cpu 

root      7588  1.0  0.2   4508  2332 ?        Ss   15:56   0:00  \_ /bin/bash /usr/bin/entry.sh /usr/local/bin/node index.js
root      7691 20.5  0.3  10664  2968 ?        D    15:56   0:11      \_ udevd
root      7704  0.1  0.3   9968  3336 ?        D    15:56   0:00      |   \_ udevd
root      7709  0.1  0.2   9972  2100 ?        D    15:56   0:00      |   \_ udevd
root      7710  0.1  0.3   9612  3248 ?        D    15:56   0:00      |   \_ udevd
root      7711  0.1  0.2   9612  2688 ?        D    15:56   0:00      |   \_ udevd
root      7712  0.1  0.2   9612  2632 ?        D    15:56   0:00      |   \_ udevd
root      7713  0.1  0.2   9612  2684 ?        D    15:56   0:00      |   \_ udevd
root      7714  0.1  0.2   9612  2444 ?        D    15:56   0:00      |   \_ udevd
root      7715  0.1  0.1   9612  1824 ?        D    15:56   0:00      |   \_ udevd
root      7721  0.1  0.1   9612  1840 ?        D    15:56   0:00      |   \_ udevd
root      7722  0.1  0.1   9612  1840 ?        D    15:56   0:00      |   \_ udevd
root      7723  0.1  0.2   9612  2400 ?        D    15:56   0:00      |   \_ udevd
root      7724  0.1  0.2   9744  2008 ?        D    15:56   0:00      |   \_ udevd
root      7725  0.1  0.2   9744  2008 ?        D    15:56   0:00      |   \_ udevd
root      7726  0.1  0.2   9744  2012 ?        D    15:56   0:00      |   \_ udevd
root      7727  0.1  0.2   9744  2072 ?        D    15:56   0:00      |   \_ udevd
root      7728  0.1  0.1   9744  1892 ?        D    15:56   0:00      |   \_ udevd

Am i alone in this situation ?

nghiant2710, 2016-11-07:
@Sid3y1, I can't reproduce your issue :/ Can you help me reproduce it?

petrosagg, 2016-11-27:
@Sid3y1 in the process list you send all the udevd threads are in disk wait state (D). This means that they are waiting for IO and therefore not using any of the CPU. Since this is a very old issue I'll close it but feel free to re-open if you have more information about the problem or ways to reproduce.

closed 2016-11-27 by petrosagg