balena-io-library/base-images #206: Consider providing yarn as part of the base image

emirotin, 2016-11-07:
https://yarnpkg.com/ is a modern npm client replacement from Facebook, Google, et. al.
Providing it pre-packaged is a matter of `npm install -g`, but should make our images look way more modern.
Would probably only make sense for `node >= 4`

Page-, 2016-11-07:
Do you happen to know how big yarn is? For the sake of the slim images I'd say we should leave it out regardless (should be bare minimum), but it'd be nice to know the cost for other images as well

nghiant2710, 2016-11-07:
there is a PR for it in official repo https://github.com/nodejs/docker-node/pull/260 but it's not merged yet. We will consider including it when it's approved in upstream repository, I think

emirotin, 2016-11-07:
OK, it's _not that small_:

```
du -sh ~/.nvm/versions/node/v6.5.0/lib/node_modules/yarn
 24M	/Users/eugene/.nvm/versions/node/v6.5.0/lib/node_modules/yarn
```

As a side not we don't have to wait for the official image change, as I said it can easily be done with a single extra `RUN` command.

nghiant2710, 2016-11-07:
I meant it's still controversial whether should it be used and how to add it into the base images. So it's better to wait a bit and see how it goes

petrosagg, 2016-11-26:
@emirotin to give some context, the reason we wait for upstream is because we want to be as aligned as possible with Docker's base images to give our users a familiar environment to work in. Granted this is a pure addition, but still an interface that we'll have to maintain if added.

Regarding the size, ouch :/

PR #373 (2018-05-02): Yarn integration
Fixes #206, yarn has been integrated into the upstream base images for a while so we should have it in our base images too.

```
root@a42bb16e1fce:/opt# du -sh yarn
3.9M    yarn
```
This is the package size on amd64 base images so I think it will no longer be an issue to us anymore.
closed 2018-05-02 by nghiant2710