balena-io-library/base-images #221: Mention in docs: default "nodoc" dpkg config may cause unexpected apt-get behavior

100ideas, 2017-01-07:
I've been developing my first resin-backed app. I started with a Dockerfiles based on `resin/raspberrypi-python` and ran it in a local docker container while I iterated on the first pass of application code - roughly the first 20% of the project.

After everything configured properly and my code mildly organized, I used resin.io to build and push the container to resin.os on a nearby raspberry pi and I switched to interactive development via ssh, with perhaps just 5-10 Dockerfile+resin.io rebuilds. The [mgrenier/remote-ftp](https://github.com/mgrenier/remote-ftp) Atom package makes remote development on the rpi super easy.

So I've ended up installing a few extra packages on my dev rpi for testing, although I intend to eventually remove them from my Dockerfile. One of them was the AWS CLI interface (`pip install awscli`), which isn't necessary in production but has been useful for quickly making sure I have everything in the pi environment needed to process various outputs from the AWS api.

[Long story short](https://github.com/aws/aws-cli/issues/1957#issuecomment-271057166), the `aws` CLI tool wasn't able to display any help files on the pi, and it took me quite a while to figure out why: because it depended on `groff`, and although `apt-get install groff` apparently produced a working `groff` binary in the PATH, it had in fact not been installed with the output modules (`ascii & utf8 devices` in groff parlance) that `aws` needed. [Because](https://github.com/resin-io-library/base-images/blob/master/debian/armv7hf/jessie/Dockerfile#L17) `apt-get` via `dpkg` had been configured in the base image to [skip installing certain kinds of documentation resources]((https://github.com/resin-io-library/base-images/blob/master/debian/armv7hf/jessie/01_nodoc)), including anything in `/usr/share/groff`. Commenting-out the line in question and `apt-get install --reinstall groff-base` solved the issue.

I think it's reasonable to exclude documentation files by default in the base images to save space, along with most of the other bloaty tools these "skinny" distros do away with. But it shouldn't lead to *unexpected* behavior of tools like `apt-get`. So please consider adding a note in the docs somewhere explaining how `dkpg` differs from the default configuration and what to expect from it.

It would be great if there were a couple of resin.os base images "fattened up" to support interactive development environments, but fundamentally identical to the existing "skinny" production images. I would 
1. do as much initial work as I could running the dev image in a local container;
2. push it over to the pi when necessary (to test embedded libraries or certain interfaces GPIO serial etc) and continue to work interactively, then bounce back to the local container;
3. verify everything worked as expected with the production base image. Maybe this would be as simple as making a single change to the `FROM` directive in the dockerfile
```Dockerfile
#FROM resin/raspberrypi-python:development
FROM resin/raspberrypi-python
```
if not, just maintaining dev and prod branches in git when the Dockerfiles required more elaborate changes.

Is anyone following a dev pattern like this already?

nghiant2710, 2017-01-11:
@100ideas thanks a lot for your feedback, we will update our docs soon.

The idea for a development base images is really cool, I'm not sure if we can emulate stuff like serial devices, GPIO in docker image but will give it a try 👍

100ideas, 2017-01-12:
Thanks - glad the feedback was useful.

Closing this comment; hopefully github search is *good enough* for doc purposes for other users that may encounter this issue.

100ideas, 2017-01-12:
Incidentally, I've switched from using `boto3` python aws library to just making calls to aws cli via `subprocess.Popen()` in python, so it has turned out handy to have the `aws help` system working on my resin rpi dev remote.

closed 2017-01-12 by 100ideas