balena-io-library/base-images #226: Failing to call CMD commands when INITSYSTEM=on

ntzovanis, 2017-01-13:
When using resin/odroid-ux3-python:latest

If on DOCKERFILE:
```
INITSYSTEM=on
CMD bash start.sh
```
then start.sh is not executed (it's only executed if called as ```ENTRYPOINT bash start.sh```)

If on DOCKERFILE:
```
INITSYSTEM=off
CMD bash start.sh
```
then start.sh is exectuted

shaunmulligan, 2017-01-13:
@ntzovanis one clarification, this is on resinOS (unmanaged) when one does `rdt push`, correct?

ntzovanis, 2017-01-13:
@shaunmulligan that's correct

shaunmulligan, 2017-01-13:
@petrosagg or @nghiant2710 any ideas on why this would happen?

nghiant2710, 2017-01-18:
@ntzovanis can you please send us the Dockerfile? I can't reproduce your issue, it seems fine on my side

shaunmulligan, 2017-01-18:
@nghiant2710 did you try reproduce this on pure resinOS with `rdt` ... the issue only appears there and not on resin.io

ntzovanis, 2017-01-18:
@shaunmulligan is right. The issue happens on pure resinOS (and rdt). I've tried it in different computers and odroids with the same result.

```FROM resin/odroid-ux3-python:latest

# Enable systemd
ENV INITSYSTEM on

ENV PYTHONUNBUFFERED=0

# Your code goes here

WORKDIR /usr/src/mqtt-client

COPY . .

RUN pip install paho-mqtt
RUN pip install requests
RUN pip install pytz

ENTRYPOINT bash start.sh```

Where start.sh can be anything and it's not called at all.

closed 2019-01-28 by nghiant2710