balena-io-library/base-images #231: Please add support for C.H.I.P. base images

bcopy, 2017-01-29:
The CHIP is a 9 dollars ARMv7 computer that is debian compatible (getchip.com) - Adding it to the list of supported platforms would be a quick win.

nghiant2710, 2017-02-07:
hey @bcopy, thanks for your suggestion, we will add the base images for that board as soon as it's supported by Resin. In the meantime, you can run the generic armv7 base images `resin/armv7hf-debian` or `resin/armhf-alpine`

rauston, 2017-02-27:
+1, @bcopy, did you get one of the generic armv7 base images to run on the C.H.I.P? If so, which one and are you using the C.H.I.P or C.H.I.P PRO board? Thanks!

bcopy, 2017-02-27:
Yeah the generic armhf images work fine.

nghiant2710, 2017-04-21:
@bcopy @rauston please follow the device support progress here https://waffle.io/resin-os/resin-os-device-support or create a request for new board here https://github.com/resin-os/resin-os-device-support.

We will create base images for it as soon as It's supported. In the meantime, you can just use resin generic armhf base images.

closed 2017-04-21 by nghiant2710