balena-io-library/base-images #233: improve tag consistency through the image hiaerarchy: add more date fixed tag

cskiraly, 2017-02-02:
I'm using images based directly on resin/raspberrypi2-debian:jessie-20161010. These images nicely share their base layers. 
For another image I need node, so I was looking for a tag in resin/raspberrypi2-node that shares the same base .... but there is no such tag, and in general I see no combination of tags that would achieve this.

I think it would be nice to be able to select date tagged images form <device-name>-<distro> and <device-name>-<distro>-node such that:
- the base layers are long term stable
- the node version only adds layers to the base version.

cskiraly, 2017-02-02:
@curcuz @craig-mulligan, can you check?

nghiant2710, 2017-02-03:
Hey @cskiraly,

You're correct, our base images follows the `<device>-<distro>-<language>` naming scheme, Debian base images are special since they are created before we defined that scheme so `debian` part is omitted.

About the tag for each base images, you should check the available tags for each images in the docker hub repository. You can find the links in our docs here https://docs.resin.io/runtime/resin-base-images/

cskiraly, 2017-02-03:
Hi @nghiant2710 ,

Even if Debian based images are outliers, I think there would be a need for date tagged images that one can consistently use at different levels, and through different languages.

nghiant2710, 2017-02-03:
@cskiraly let me explain about how we build the base images.

The Resin base images are automatically rebuilt every week to make sure all images are up-to-date but there is a problem here, rebuilding base images can potentially break user's builds. To solve that problem, we create the date tagged images, which will never be rebuilt, after every successful build (we don't schedule the date, it's automatically handled by our builders), so users can base their app on that image and it will be stable.

If you want to find node image which uses a specify base layer, you should look for the nearest date in tag list on dockerhub. For example: if you want node image of `raspberrypi2-debian:jessie-20161010`, it's `resin/raspberrypi2-node:6-20161022` (this tag is too far from now so dockerhub doesn't display the tag but I found it in our builders logs)

Also we do have a PR to export all tags to a github repository so you can easily find all available tags because dockerhub doesn't display all tags.

shaunmulligan, 2017-02-03:
@petrosagg and @nghiant2710 surely there must be some way to only tag all the builds for a particular build date in one step so that they all have exactly the same date tag?

cskiraly, 2017-02-03:
Thanks @nghiant2710, I wasn't aware DockerHub limits the list of tags shown! Is there any simple way to list all available tags?
If possible, I would still vote for having tags date synched.

cskiraly, 2018-04-10:
Since the resinOS model now changed to multi-container, let me revisit this issue reiterating my suggestion:

The problem:
When preparing a set of containers to be deployed on a device, one could try to optimize the size of the overall distribution by generating images with common base layers. However, it is still cumbersome to rely on image timestamps, since the timestamp is the date of generation, and the generation process runs for several days. Thus, is it hard to find different images from the hierarchy with matching base layers, which would reduces the overall multi-container distribution size. 

My suggestion is to tag your images with 2 tags:
 - one based on the generation date, like now;
 - another one based on the date when the build cycle started.
This latter would be a tag consistent over the whole image hierarchy, largely simplifying automation of the creation of consistent multi-image sets.

@shaunmulligan as you can see I've already opened an issue on this

nghiant2710, 2019-12-28:
hey @cskiraly,

This issue is resolved in the `balenalib` base images as the datestamp is now consistent. All base images generated from the same release will now have same datestamp.

closed 2019-12-28 by nghiant2710