balena-io-library/base-images #240: Can't launch bash with Image resin/ts7700-buildpack-deps:jessie-20170222

bjet007, 2017-02-24:
Hi 

I'm trying to run the following docker image on mac os and it fail without any exception other than an exit code 149:
`docker run -ti resin/ts7700-buildpack-deps:jessie-20170222 bash` 

The same command with will work with version jessie-20170201.  It seem that all images after that one are not working.

docker version: Docker version 1.13.1, build 092cba3
os version : Darwin Kernel Version 15.6.0: Mon Jan  9 23:07:29 PST 2017; root:xnu-3248.60.11.2.1~1/RELEASE_X86_64 x86_64

thanks

Benoit

nghiant2710, 2017-02-27:
hey @bjet007, we have a few changes in the entryscript that caused this issue. We will fix this issue soon and in the meantime, please use the `jessie-20170201` tag instead.

We will let you know as soon as it's fixed.

nghiant2710, 2017-03-31:
fixed in https://github.com/resin-io-library/base-images/pull/238

closed 2017-03-31 by nghiant2710