balena-io-library/base-images #254: /bin/bash not starting up in some images

cskiraly, 2017-03-23:
The following command silently fails for me:
`docker run -it resin/raspberrypi3-openjdk:openjdk-8-jdk-20170217 /bin/bash`

The same works with other images, e.g. this works:
`docker run -it resin/raspberrypi2-openjdk:openjdk-7-jdk /bin/bash`

In the case of failure, with `docker events` I see this:
`2017-03-23T11:21:17.681294972+01:00 container die 138dbbcc6cfc5e71fc4406a37205234b11782292d9c0e213575b3909a708c63c (exitCode=149, image=resin/raspberrypi3-openjdk:openjdk-8-jdk-20170217, io.resin.architecture=armv7hf, io.resin.device-type=raspberrypi2, io.resin.qemu.version=2.7.0-resin-rc2-arm, name=compassionate_volhard)`

Any ideas what could be the difference between the images that make it fail?

nghiant2710, 2017-03-27:
hey @cskiraly, there was a change in the entryscript that breaks the interactive shell, we will have a fix for it soon. At the moment, please use the `openjdk-8-jdk-20170217` tag or any tag on the same date

nghiant2710, 2017-03-31:
Fixed in https://github.com/resin-io-library/base-images/pull/238

closed 2017-03-31 by nghiant2710