balena-io-library/base-images #273: Errors installing OpenJDK due to unexistent man pages directory

jviotti, 2017-05-03:
```
[Build]    aspell-autobuildhash: processing: en [en_US-wo_accents-only].
[Build]    Processing triggers for ca-certificates (20141019+deb8u2) ...
[Build]    Updating certificates in /etc/ssl/certs... 0 added, 0 removed; done.
[Build]    Running hooks in /etc/ca-certificates/update.d....
[Build]    done.
[Build]    done.
[Build]    Processing triggers for menu (2.1.47) ...
[Build]    Errors were encountered while processing:
[Build]     openjdk-7-jre-headless:amd64
[Build]     openjdk-7-jre:amd64
[Build]     default-jre
[Build]     arduino
[Build]    E: Sub-process /usr/bin/dpkg returned an error code (1)
[Build]    Removing intermediate container f6f6f937a059
[Error] Build failed: The command '/bin/sh -c apt-get update && apt-get install -y --no-install-recommends xorg mono-complete libasound2-dev fluxbox dbus arduino g++ gcc usbutils make && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 100
```

And the root cause is:

```
[Build] update-alternatives: error: error creating symbolic link `/usr/share/man/man1/rmid.1.gz.dpkg-tmp': No such file or directory
```

The `/usr/share/man/man1` directory doesn't exist, and simply doing `mkdir /usr/share/man/man1` before installing OpenJDK fixes the issue.

Maybe this is something we can do by default?

bakavic, 2017-05-04:
I'm also experiencing this problem with installing postgres - `/usr/share/man/` directory seems to be expected by a lot of config scripts? This is despite [01_nodoc](https://github.com/resin-io-library/base-images/blob/095dc766991af2f98e23655fa8ca492d5e2df1dc/debian/01_nodoc), which is suppose to exclude these files.

```
Setting up postgresql-client-9.4 (9.4.10-0+deb8u1) ...
[34m[Build]   [39m update-alternatives: using /usr/share/postgresql/9.4/man/man1/psql.1.gz to provide /usr/share/man/man1/psql.1.gz (psql.1.gz) in auto mode
[34m[Build]   [39m update-alternatives: error: error creating symbolic link `/usr/share/man/man7/ALTER_EVENT_TRIGGER.7.gz.dpkg-tmp': No such file or directory
```

Seems to be due to the debian *-slim images not including this directory (https://github.com/tianon/docker-brew-debian/issues/48), and resin.io using it in https://github.com/resin-io-library/base-images/issues/265. And that's when our problems begin.

I think this is a breaking issue, and an empty `man` directory should be created.

closed 2017-06-12 by nghiant2710

bor8, 2017-12-06:
With me it was also necessary to put on man7: `mkdir /usr/share/man/man7`

commit a56e1e5b (2017-06-12): Fix #273, Create empty man directory since it fails to install many packages due to unexistent man pages directory
commit fce1ef6d (2017-06-12): Merge pull request #276 from nghiant2710/273-mising-manpage-dir

Fix #273, Create empty man directory to fix missing manpage directory issue