balena-io-library/base-images #297: qemu: Unsupported syscall: 370

TheOddler, 2017-06-24:
Hi,

When I try to run the `resin/raspberrypi3-python:3-onbuild` image on my Windows 10 pc (will of course later be run on a Raspberry Pi), I get the error `Unsupported syscall: 370` several times, 7 times to be exact.

This is the output I get:

```
Systemd init system enabled.
qemu: Unsupported syscall: 370
qemu: Unsupported syscall: 370
qemu: Unsupported syscall: 370
qemu: Unsupported syscall: 370
qemu: Unsupported syscall: 370
qemu: Unsupported syscall: 370
qemu: Unsupported syscall: 370
```

Then it waits, and continue running, it doesn't quit automatically, I have to stop it using `Docker stop`.

I was trying out this image for a simple flask app I'm making, but I even get the error on an clean project. There are only 2 files in the clean project, my Dockerfile and an empty `requirements.txt`.

The docker file looks like this:

```
FROM resin/raspberrypi3-python:3-onbuild

ENV INITSYSTEM on
```

To build the project I used:

    docker build -t theoddler/pipyresintest .

and then to run it:

    docker run theoddler/pipyresintest

Any help why this happens? Am I using the image wrong?

TheOddler, 2017-06-24:
I just tried running docker with the `-it` options, and now I get an additional error:

```
PS C:\Users\Pablo\Dev\Gnodes\Code\Web_Hub_V0> docker run -it --rm --name gnodes
theoddler/gnodes
Systemd init system enabled.
qemu: Unsupported syscall: 370
qemu: Unsupported syscall: 370
qemu: Unsupported syscall: 370
qemu: Unsupported syscall: 370
qemu: Unsupported syscall: 370
qemu: Unsupported syscall: 370
qemu: Unsupported syscall: 370
Failed to mount tmpfs at /run: Operation not permitted
```

TheOddler, 2017-06-24:
I now tried running this on my pi, with privileges. However there nothing happens, it runs without any output and then quits.

When I enable `events` first (`docker events&`) and then run the image, it seems to create and start the container fine, but then immediatly dies :(

```
pi@raspberrypi:~ $ docker run --privileged -it theoddler/gnodes
2017-06-24T11:13:09.500711722Z container create 06ef0691682b2a0c42e5052316168a0b3ccc53054c043013e28a9ebdb83ce53e (image=theoddler/gnodes, io.resin.architecture=armv7hf, io.resin.device-type=raspberry-pi2, io.resin.qemu.version=2.9.0.resin1-arm, name=blissful_goldwasser)
2017-06-24T11:13:09.538316469Z container attach 06ef0691682b2a0c42e5052316168a0b3ccc53054c043013e28a9ebdb83ce53e (image=theoddler/gnodes, io.resin.architecture=armv7hf, io.resin.device-type=raspberry-pi2, io.resin.qemu.version=2.9.0.resin1-arm, name=blissful_goldwasser)
2017-06-24T11:13:11.585910255Z network connect a421442b7614c0f967a776fd9c113cdad9b96c5f30869fbf83364a0eb1f479bd (container=06ef0691682b2a0c42e5052316168a0b3ccc53054c043013e28a9ebdb83ce53e, name=bridge, type=bridge)
2017-06-24T11:13:11.723163683Z volume mount a2fa968d4aaf505234e8c4a9bab34270a21ae2ea5fd4b21a65f57ac05ba5b139 (container=06ef0691682b2a0c42e5052316168a0b3ccc53054c043013e28a9ebdb83ce53e, destination=/sys/fs/cgroup, driver=local, propagation=, read/write=true)
2017-06-24T11:13:17.130655548Z container start 06ef0691682b2a0c42e5052316168a0b3ccc53054c043013e28a9ebdb83ce53e (image=theoddler/gnodes, io.resin.architecture=armv7hf, io.resin.device-type=raspberry-pi2, io.resin.qemu.version=2.9.0.resin1-arm, name=blissful_goldwasser)
2017-06-24T11:13:17.220043571Z container die 06ef0691682b2a0c42e5052316168a0b3ccc53054c043013e28a9ebdb83ce53e (exitCode=139, image=theoddler/gnodes, io.resin.architecture=armv7hf, io.resin.device-type=raspberry-pi2, io.resin.qemu.version=2.9.0.resin1-arm, name=blissful_goldwasser)
2017-06-24T11:13:18.518302326Z network disconnect a421442b7614c0f967a776fd9c113cdad9b96c5f30869fbf83364a0eb1f479bd (container=06ef0691682b2a0c42e5052316168a0b3ccc53054c043013e28a9ebdb83ce53e, name=bridge, type=bridge)
2017-06-24T11:13:18.547163365Z volume unmount a2fa968d4aaf505234e8c4a9bab34270a21ae2ea5fd4b21a65f57ac05ba5b139 (container=06ef0691682b2a0c42e5052316168a0b3ccc53054c043013e28a9ebdb83ce53e, driver=local)
```

nghiant2710, 2017-06-24:
@TheOddler Sorry for the late response. About running the container on Windows 10, you shouldn't enable systemd since QEMU can't emulate all the calls from systemd therefore there are few `unsupported syscall` logs, just remove `ENV INITSYSTEM on` from your Dockerfile and it will be fine.

Regarding the second question, what is your Dockerfile for `theoddler/gnodes` image? is it the same with `theoddler/pipyresintest`. Also have you tried to run `docker logs` to fetch the logs of a container?

TheOddler, 2017-06-24:
@nghiant2710 Thanks for the reply.

First: Should I not-enable the systemd only on windows? Or not on the Pi either? And when should I enable it? The description on dockerhub for the base image (https://hub.docker.com/r/resin/raspberry-pi2-python/) and a quick google (https://docs.resin.io/runtime/runtime/) seemed to suggest I should just enable it always, or risk loosing valuable information for debugging when something goes wrong.

My Dockerfile for the gnodes project is:

```
# Pick a base image (onbuild adds triggers for building the image, it copies the python files, installs dependencies, ...)
FROM resin/raspberry-pi2-python:3-onbuild

# Enable container init system. See https://docs.resin.io/runtime/runtime/
ENV INITSYSTEM on

# Expose the 8080 port, which my app runs on
EXPOSE 8080

# A command for starting my app, this is run at the start of a container
CMD ["python", "./app.py"]
```

So basically the same, but exposes the port my app uses, and starts the flask app.

I get similar output for the `theoddler/pipyresintest` image:

```
pi@raspberrypi:~ $ docker run -it --privileged=true -p 8080:8080 --rm theoddler/pipyresinte
st
2017-06-24T12:26:35.576459929Z container create 83a3401ab1b01a05e092467df6ade756a4dfb99235e21443cbad2d54f248ee14 (image=theoddler/pipyresintest, io.resin.architecture=armv7hf, io.resin.device-type=raspberry-pi2, io.resin.qemu.version=2.9.0.resin1-arm, name=stupefied_bardeen)
2017-06-24T12:26:35.614876666Z container attach 83a3401ab1b01a05e092467df6ade756a4dfb99235e21443cbad2d54f248ee14 (image=theoddler/pipyresintest, io.resin.architecture=armv7hf, io.resin.device-type=raspberry-pi2, io.resin.qemu.version=2.9.0.resin1-arm, name=stupefied_bardeen)
2017-06-24T12:26:37.880433152Z network connect a421442b7614c0f967a776fd9c113cdad9b96c5f30869fbf83364a0eb1f479bd (container=83a3401ab1b01a05e092467df6ade756a4dfb99235e21443cbad2d54f248ee14, name=bridge, type=bridge)
2017-06-24T12:26:38.022000496Z volume mount 53331011523fa44de2dd75963d6326483e5cbc570e787f3173832ea277a43368 (container=83a3401ab1b01a05e092467df6ade756a4dfb99235e21443cbad2d54f248ee14, destination=/sys/fs/cgroup, driver=local, propagation=, read/write=true)
2017-06-24T12:26:42.307696539Z container start 83a3401ab1b01a05e092467df6ade756a4dfb99235e21443cbad2d54f248ee14 (image=theoddler/pipyresintest, io.resin.architecture=armv7hf, io.resin.device-type=raspberry-pi2, io.resin.qemu.version=2.9.0.resin1-arm, name=stupefied_bardeen)
2017-06-24T12:26:42.373345380Z container die 83a3401ab1b01a05e092467df6ade756a4dfb99235e21443cbad2d54f248ee14 (exitCode=139, image=theoddler/pipyresintest, io.resin.architecture=armv7hf, io.resin.device-type=raspberry-pi2, io.resin.qemu.version=2.9.0.resin1-arm, name=stupefied_bardeen)
2017-06-24T12:26:44.118281989Z network disconnect a421442b7614c0f967a776fd9c113cdad9b96c5f30869fbf83364a0eb1f479bd (container=83a3401ab1b01a05e092467df6ade756a4dfb99235e21443cbad2d54f248ee14, name=bridge, type=bridge)
2017-06-24T12:26:44.172251214Z volume unmount 53331011523fa44de2dd75963d6326483e5cbc570e787f3173832ea277a43368 (container=83a3401ab1b01a05e092467df6ade756a4dfb99235e21443cbad2d54f248ee14, driver=local)
2017-06-24T12:26:46.164121701Z container destroy 83a3401ab1b01a05e092467df6ade756a4dfb99235e21443cbad2d54f248ee14 (image=theoddler/pipyresintest, io.resin.architecture=armv7hf, io.resin.device-type=raspberry-pi2, io.resin.qemu.version=2.9.0.resin1-arm, name=stupefied_bardeen)
```

But there it would make sense, since it doesn't start any app, so I would expect Docker to quit it quite fast. Right?

nghiant2710, 2017-06-24:
@TheOddler, you shouldn't enable systemd on windows 10 and can enable it on the Pi (arm environment). However for debugging purpose, I think you should disable systemd then rebuild the Dockerfile and run `docker run -it --rm --name gnodes theoddler/gnodes bash`, it will start an interactive shell in the container, then you can manually run `python app.py` and see what happens to your app.

TheOddler, 2017-06-24:
@nghiant2710 Thanks!

I'll try that next, if my current test doesn't work. I edited my previous answer, aparently the Pi I"m using is a Pi 1, rather than a Pi 2. I just grabbed one I had lying around, assuming it was a 2, but that was a wrong assumption. So I'm rebuilding the image now with the Raspberry Pi 1 base image.

TheOddler, 2017-06-24:
Ok, rebuilding the project with the proper base image worked. It's now running, and seems to work. Removing 'ENV INITSYSTEM on' did indeed remove the errors.

Thanks for the help, and sorry for wasting your time by using the wrong base image. Should have properly checken my pi, rather than assuming it would be a Pi 2.

closed 2017-06-24 by TheOddler

TheOddler, 2017-06-25:
@nghiant2710 This might not be the best place to ask this, but I was wondering, is there an image that supports all raspberry pi's?