balena-io-library/base-images #299: Create ZuluJDK variant for Java images

petrosagg, 2017-07-02:
On some platforms ZuluJDK has superior performance:

```
Native build performance on RPi3 + Samsung EVO+ 32GB SD:
  mvn package of recommenderandconfigurator
  - with OpenJDK:                                            10 minutes
  - with ZuluJDK in default mixed mode:            1 minutes
  - with ZuluJDK in interpreted mode (-Xint):  5.5 minutes

Runtime performance
  Spring Boot startup times for the recommenderandconfigurator service
  - with OpenJDK:                                            180 seconds
  - with ZuluJDK in default mixed mode:            13.5 seconds
  - with ZuluJDK in interpreted mode (-Xint):     66.5 seconds
```

cskiraly, 2017-07-03:
The above test results are reproducible based on https://github.com/Agile-IoT/Configurator/tree/jdk-perftest. 

More specifically:
https://github.com/Agile-IoT/Configurator/tree/7c160429e80a65223ef4a34200b6e041818c2296
and/or the following docker images:
agileiot/recommenderandconfigurator:perf-openjdk
agileiot/recommenderandconfigurator:perf-zulujdk
agileiot/recommenderandconfigurator:perf-zulujdk-Xint

cskiraly, 2017-07-05:
I've done some other ad-hoc testing (to be repeated with more care).

- Maven download: 
  - capped at 40 KB/s with the OpenJDK build
  - no cap or cap high with ZuluJDK build

- Triggering an HTTP endpoint (Jetty behind): 
  - 1 tps with the OpenJDK build
  - 20 tps with ZuluJDK build

commit 1dcb555a (2017-08-28): Merge pull request #315 from nghiant2710/openjdk

Fix #299, add support for zulu variant openjdk base images
closed 2017-08-28 by nghiant2710