balena-io-library/base-images #317: Environment variables with dollar signs break on alpine with initsystem on

petrosagg, 2017-08-01:
On Alpine with `INITSYSTEM=on` env vars with dollar signs in their value like `FOO=$bar$baz` will be interpreted before being passed to the application.

This is probably where the error happens where variables are not escaped with `%q`
https://github.com/resin-io-library/base-images/blob/master/alpine/entry.sh#L38-L51

nghiant2710, 2017-08-02:
it's not the problem with `%s`. In my test app, it passes correct variable value to the openrc init script (`$Foo`).
```
start() {
        ebegin "Starting Resin.io User Application"
        start-stop-daemon --start --exec /etc/resinApp.sh \
        --chdir / \
        --env "RESIN_SUPERVISOR_API_KEY=8700b0cb0f53958096f7be683757b4d0286d1597912d5d91099fc387a0720bfb" --env "HOSTNAME=e1a0aef" --env "RESIN_APP_ID=2211" --env "TERM=xterm" --env "RESIN=1" --env "RESIN_DEVICE_TYPE=raspberry-pi" --env "RESIN_SUPERVISOR_ADDRESS=http://127.0.0.1:48484" --env "USER=root" --env "RESIN_SUPERVISOR_HOST=127.0.0.1" --env "RESIN_DEVICE_UUID=e1a0aef432a52daa55cce334557a24b3" --env "PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" --env "RESIN_API_KEY=0a461569e17924c1812275ed4c7f9a0a" --env "PWD=/" --env "RESIN_SUPERVISOR_VERSION=4.2.2" --env "RESIN_APP_NAME=avahi" --env "INITSYSTEM=on" --env "Foo=$2a$08$p7SNNZKTCvepTYQRxzwG0O5cVpWgoiILViIaFd7XW7FbWVcGacqMu" --env "RESIN_DEVICE_NAME_AT_INIT=bitter-dew" --env "SHLVL=1" --env "HOME=/root" --env "RESIN_HOST_OS_VERSION=Resin" --env "OS" --env "2.0.3+rev1" --env "RESIN_SUPERVISOR_PORT=48484" --env "_=/usr/bin/env"  \
        --make-pidfile --pidfile "$pid_file" --background \
        --stdout /dev/console \
        --stderr /dev/console \
        --wait "$waiting_time"
        eend $?
}
```

nghiant2710, 2017-08-02:
This is the problem of double quote, it expands dollar signs as variables. Changing to single quote will fix the issue, tested with some special characters and it works.
```
02.08.17 17:33:27 (+0700)    OpenRC 0.24.1.e4cc4b79b5 is starting up Linux 4.4.50 (armv6l) [LXC]
02.08.17 17:33:27 (+0700)
02.08.17 17:33:27 (+0700)  * /proc is already mounted
02.08.17 17:33:27 (+0700)  * Mounting /run ...
02.08.17 17:33:27 (+0700)  * /run/openrc: creating directory
02.08.17 17:33:27 (+0700)  * /run/lock: creating directory
02.08.17 17:33:27 (+0700)  * /run/lock: correcting owner
02.08.17 17:33:27 (+0700)  * Caching service dependencies ...
02.08.17 17:33:28 (+0700) Started application 'registry2.resin.io/avahi/67ce2082b59557205699591050247c9afb2fb73e'
02.08.17 17:33:29 (+0700)  [ ok ]
02.08.17 17:33:29 (+0700)  * Starting D-BUS system messagebus ...
02.08.17 17:33:31 (+0700)  [ ok ]
02.08.17 17:33:32 (+0700)  * Starting Resin.io User Application ...
02.08.17 17:33:33 (+0700)  [ ok ]
02.08.17 17:33:34 (+0700) $2a$08$p7'@"SNNZKTCvepTYQRxzwG0O5cVpWgoiILViIaFd7XW7FbWVcGacqMu
02.08.17 17:33:34 (+0700) RESIN_SUPERVISOR_API_KEY=8700b0cb0f53958096f7be683757b4d0286d1597912d5d91099fc387a0720bfb
02.08.17 17:33:34 (+0700) HOSTNAME=e1a0aef
02.08.17 17:33:34 (+0700) RESIN_APP_ID=2211
02.08.17 17:33:34 (+0700) TERM=xterm
02.08.17 17:33:34 (+0700) SHELL=/bin/sh
02.08.17 17:33:34 (+0700) FOO=
02.08.17 17:33:34 (+0700) EINFO_LOG=/etc/init.d/resin
02.08.17 17:33:34 (+0700) RESIN=1
02.08.17 17:33:34 (+0700) RESIN_DEVICE_TYPE=raspberry-pi
02.08.17 17:33:34 (+0700) RESIN_SUPERVISOR_ADDRESS=http://127.0.0.1:48484
02.08.17 17:33:34 (+0700) USER=root
02.08.17 17:33:34 (+0700) RESIN_SUPERVISOR_HOST=127.0.0.1
02.08.17 17:33:34 (+0700) RESIN_DEVICE_UUID=e1a0aef432a52daa55cce334557a24b3
02.08.17 17:33:34 (+0700) RC_SVCNAME=resin
02.08.17 17:33:34 (+0700) PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
02.08.17 17:33:34 (+0700) RESIN_API_KEY=0a461569e17924c1812275ed4c7f9a0a
02.08.17 17:33:34 (+0700) RUNLEVEL=quiet
02.08.17 17:33:34 (+0700) RESIN_SUPERVISOR_VERSION=4.2.2
02.08.17 17:33:34 (+0700) RESIN_APP_RELEASE=67ce2082b59557205699591050247c9afb2fb73e
02.08.17 17:33:34 (+0700) PWD=/
02.08.17 17:33:34 (+0700) RESIN_APP_NAME=avahi
02.08.17 17:33:34 (+0700) INITSYSTEM=on
02.08.17 17:33:34 (+0700) FOO1=$2a$08$p7'@"SNNZKTCvepTYQRxzwG0O5cVpWgoiILViIaFd7XW7FbWVcGacqMu
02.08.17 17:33:34 (+0700) RESIN_DEVICE_NAME_AT_INIT=bitter-dew
02.08.17 17:33:34 (+0700) SVCNAME=resin
02.08.17 17:33:34 (+0700) SHLVL=2
02.08.17 17:33:34 (+0700) HOME=/
02.08.17 17:33:34 (+0700) RESIN_HOST_OS_VERSION=Resin OS 2.0.3+rev1
02.08.17 17:33:34 (+0700) RC_SERVICE=/etc/init.d/resin
02.08.17 17:33:34 (+0700) RESIN_SUPERVISOR_PORT=48484
02.08.17 17:33:34 (+0700) _=/usr/bin/env
```

nghiant2710, 2017-08-02:
actually we don't need to fix it that way, the reason we need to manually set all variables is this option https://github.com/resin-io-library/base-images/blob/master/alpine/rc.conf#L4 didn't work but on the new Alpine Linux versions (3.5 and 3.6) with OpenRC (v0.21 and v0.24 respectively), it works. Since we only supports Alpine Linux v3.5 and v3.6 now so just need to drop environment replacing and we're good.

commit 297a603c (2017-08-07): Fix #317, Fix issue when value contains dollar signs will be interpreted before being passed to the application on Alpine Linux
commit 827cf9fb (2017-08-07): Merge pull request #318 from nghiant2710/alpine

Fix #317, Fix issue when value contains dollar signs will be interpreted before being passed to the application on Alpine Linux
closed 2017-08-07 by nghiant2710