balena-io-library/base-images #319: Missing /sbin/init in Debian Stretch

imrehg, 2017-08-04:
Using `raspberrypi3-debian:stretch` with `ENV INITSYSTEM=on`, but the application starts like this:

```
04.08.17 10:37:42 (+0100) Systemd init system enabled.
04.08.17 10:37:43 (+0100) /usr/bin/entry.sh: line 80: /sbin/init: No such file or directory
04.08.17 10:37:44 (+0100) Systemd init system enabled.
04.08.17 10:37:44 (+0100) /usr/bin/entry.sh: line 80: /sbin/init: No such file or directory
04.08.17 10:37:46 (+0100) Systemd init system enabled.
04.08.17 10:37:46 (+0100) /usr/bin/entry.sh: line 80: /sbin/init: No such file or directory
04.08.17 10:37:48 (+0100) Systemd init system enabled.
04.08.17 10:37:49 (+0100) /usr/bin/entry.sh: line 80: /sbin/init: No such file or directory
...
```

The same code works on  `raspberrypi3-debian:jessie` right now.

closed 2017-08-28 by nghiant2710

nmaas87, 2017-09-18:
Same problem still is in
resin/rpi-raspbian:stretch - or is this now deprecated?

nmaas87, 2017-09-18:
````
root@resin:~# docker run -i -t resin/rpi-raspbian:stretch /bin/bash
root@9b2f60bd6e4f:/#
root@9b2f60bd6e4f:/# which init
root@9b2f60bd6e4f:/#
root@9b2f60bd6e4f:/# ls /sbin
MAKEDEV        cfdisk      dumpe2fs  fsck         fstab-decode   isosize   mkfs         mkhomedir_helper  pam_tally2  sfdisk             switch_root  wipefs
agetty         chcpu       e2fsck    fsck.cramfs  fstrim         killall5  mkfs.bfs     mkswap            pivot_root  shadowconfig       sysctl       zramctl
badblocks      ctrlaltdel  e2image   fsck.ext2    getty          ldconfig  mkfs.cramfs  mntctl            plymouthd   start-stop-daemon  tini
blkdeactivate  debugfs     e2label   fsck.ext3    hwclock        logsave   mkfs.ext2    modinfo           raw         sulogin            tune2fs
blkdiscard     depmod      e2undo    fsck.ext4    initctl        losetup   mkfs.ext3    modprobe          resize2fs   swaplabel          udevadm
blkid          dmsetup     fdisk     fsck.minix   insmod         lsmod     mkfs.ext4    mountall          rmmod       swapoff            unix_chkpwd
blockdev       dmstats     findfs    fsfreeze     installkernel  mke2fs    mkfs.minix   pam_tally         runuser     swapon             unix_update
root@9b2f60bd6e4f:/#
````

nghiant2710, 2017-09-19:
@nmaas87 after installing `systemd-sysv` package, you will see `/sbin/init` there. However there is a problem with systemd on stretch and docker (ref https://github.com/resin-io-library/base-images/issues/323) so if you need systemd, you will have to change to jessie atm.

PR #324 (2017-08-28): Fix broken systemd in Debian Stretch
Fix #319 and #323. Details mentioned in issue #323.
Downgrade systemd to v230-7(jessie-backports) to make it works again.

This PR only fixes systemd on Debian Stretch. On Fedora 25, I'm able to downgrade systemd to v229 (Fedora 24) but it doesn't work properly yet so the fix for Fedora will be submitted later.