balena-io-library/base-images #331: Go 1.9 support

bgentry, 2017-08-24:
Hi there,

[Go 1.9 was just released](https://blog.golang.org/go1.9). I attempted to PR this by adding the following to the official distribution checksums:

```
d70eadefce8e160638a9a6db97f7192d8463069ab33138893ad3bf31b0650a79  go1.9.linux-amd64.tar.gz
7cccff99dacf59162cd67f5b11070d667691397fd421b0a9ad287da019debc4f  go1.9.linux-386.tar.gz
```

Unfortunately, the `generate_dockerfile.sh` script breaks on my Mac:

```
+ for goVersion in '$goVersions'
++ expr match 1.4.3 '\([0-9]*\.[0-9]*\)'
expr: syntax error
```

The `generate-stackbrew-library.sh` script also breaks for me.

I also realized that I won't be able to generate the Resin in-house binary checksums, so I'm assuming this update would need to be done by somebody on the Resin team? Thanks 🙏

nghiant2710, 2017-08-25:
Thanks a lot for reminding us about Go 1.9 release. It will be added soon.

About building in-house binaries, you should send the pull request to https://github.com/resin-io/golang-arm (We will move the repo to resin-io-library account soon)

nghiant2710, 2017-10-09:
Go 1.9 added https://github.com/resin-io-library/base-images/pull/334

closed 2017-10-09 by nghiant2710