balena-io-library/base-images #332: ts7700 golang image doesn't work

ensonik, 2017-08-29:
Hi,

I've been trying to get the ts7700 go lang docker images to run and most fail with the following error. 

Wondering if you have seen this or have a workaround.

(I've tried on both a Mac and Ubuntu box and both fail with the same error)

Thanks!

`docker run -ti --rm resin/ts7700-golang:1.8 go get github.com/asciimoo/wuzz`

```
[WARN  tini (18)] Tini is not running as PID 1 and isn't registered as a child subreaper.
Zombie processes will not be re-parented to Tini, so zombie reaping won't work.
To fix the problem, use the -s option or set the environment variable TINI_SUBREAPER to register Tini as a child subreaper, or run Tini as PID 1.
# github.com/fatih/color/vendor/github.com/mattn/go-colorable
runtime: failed to create new OS thread (have 2 already; errno=22)
fatal error: newosproc

runtime stack:
runtime.throw(0x612dd8, 0x9)
	/go/src/runtime/panic.go:596 +0x70
runtime.newosproc(0x10c26000, 0x10c36000)
	/go/src/runtime/os_linux.go:163 +0x158
runtime.newm(0x62d648, 0x0)
	/go/src/runtime/proc.go:1628 +0x10c
runtime.main.func1()
	/go/src/runtime/proc.go:126 +0x24
runtime.systemstack(0x8b6800)
	/go/src/runtime/asm_arm.s:264 +0x80
runtime.mstart()
	/go/src/runtime/proc.go:1132

goroutine 1 [running]:
runtime.systemstack_switch()
	/go/src/runtime/asm_arm.s:209 +0x4 fp=0x10c227bc sp=0x10c227b8
runtime.main()
	/go/src/runtime/proc.go:127 +0x58 fp=0x10c227e4 sp=0x10c227bc
runtime.goexit()
	/go/src/runtime/asm_arm.s:1017 +0x4 fp=0x10c227e4 sp=0x10c227e4
# github.com/mitchellh/go-homedir
runtime: failed to create new OS thread (have 2 already; errno=22)
fatal error: newosproc

runtime stack:
runtime.throw(0x612dd8, 0x9)
	/go/src/runtime/panic.go:596 +0x70
runtime.newosproc(0x10c26000, 0x10c36000)
	/go/src/runtime/os_linux.go:163 +0x158
runtime.newm(0x62d648, 0x0)
	/go/src/runtime/proc.go:1628 +0x10c
runtime.main.func1()
	/go/src/runtime/proc.go:126 +0x24
runtime.systemstack(0x8b6800)
	/go/src/runtime/asm_arm.s:264 +0x80
runtime.mstart()
	/go/src/runtime/proc.go:1132

goroutine 1 [running]:
runtime.systemstack_switch()
	/go/src/runtime/asm_arm.s:209 +0x4 fp=0x10c227bc sp=0x10c227b8
runtime.main()
	/go/src/runtime/proc.go:127 +0x58 fp=0x10c227e4 sp=0x10c227bc
runtime.goexit()
	/go/src/runtime/asm_arm.s:1017 +0x4 fp=0x10c227e4 sp=0x10c227e4
# github.com/fatih/color/vendor/github.com/mattn/go-isatty
runtime: failed to create new OS thread (have 2 already; errno=22)
fatal error: newosproc

runtime stack:
runtime.throw(0x612dd8, 0x9)
	/go/src/runtime/panic.go:596 +0x70
runtime.newosproc(0x10c26000, 0x10c36000)
	/go/src/runtime/os_linux.go:163 +0x158
runtime.newm(0x62d648, 0x0)
	/go/src/runtime/proc.go:1628 +0x10c
runtime.main.func1()
	/go/src/runtime/proc.go:126 +0x24
runtime.systemstack(0x8b6800)
	/go/src/runtime/asm_arm.s:264 +0x80
runtime.mstart()
	/go/src/runtime/proc.go:1132

goroutine 1 [running]:
runtime.systemstack_switch()
	/go/src/runtime/asm_arm.s:209 +0x4 fp=0x10c227bc sp=0x10c227b8
runtime.main()
	/go/src/runtime/proc.go:127 +0x58 fp=0x10c227e4 sp=0x10c227bc
runtime.goexit()
	/go/src/runtime/asm_arm.s:1017 +0x4 fp=0x10c227e4 sp=0x10c227e4
# github.com/BurntSushi/toml
runtime: failed to create new OS thread (have 2 already; errno=22)
fatal error: newosproc

runtime stack:
runtime.throw(0x612dd8, 0x9)
	/go/src/runtime/panic.go:596 +0x70
runtime.newosproc(0x10c26000, 0x10c36000)
	/go/src/runtime/os_linux.go:163 +0x158
runtime.newm(0x62d648, 0x0)
	/go/src/runtime/proc.go:1628 +0x10c
runtime.main.func1()
	/go/src/runtime/proc.go:126 +0x24
runtime.systemstack(0x8b6800)
	/go/src/runtime/asm_arm.s:264 +0x80
runtime.mstart()
	/go/src/runtime/proc.go:1132

goroutine 1 [running]:
runtime.systemstack_switch()
	/go/src/runtime/asm_arm.s:209 +0x4 fp=0x10c227bc sp=0x10c227b8
runtime.main()
	/go/src/runtime/proc.go:127 +0x58 fp=0x10c227e4 sp=0x10c227bc
runtime.goexit()
	/go/src/runtime/asm_arm.s:1017 +0x4 fp=0x10c227e4 sp=0x10c227e4
# github.com/tidwall/match
runtime: failed to create new OS thread (have 2 already; errno=22)
fatal error: newosproc

runtime stack:
runtime.throw(0x612dd8, 0x9)
	/go/src/runtime/panic.go:596 +0x70
runtime.newosproc(0x10c26000, 0x10c36000)
	/go/src/runtime/os_linux.go:163 +0x158
runtime.newm(0x62d648, 0x0)
	/go/src/runtime/proc.go:1628 +0x10c
runtime.main.func1()
	/go/src/runtime/proc.go:126 +0x24
runtime.systemstack(0x8b6800)
	/go/src/runtime/asm_arm.s:264 +0x80
runtime.mstart()
	/go/src/runtime/proc.go:1132

goroutine 1 [running]:
runtime.systemstack_switch()
	/go/src/runtime/asm_arm.s:209 +0x4 fp=0x10c227bc sp=0x10c227b8
runtime.main()
	/go/src/runtime/proc.go:127 +0x58 fp=0x10c227e4 sp=0x10c227bc
runtime.goexit()
	/go/src/runtime/asm_arm.s:1017 +0x4 fp=0x10c227e4 sp=0x10c227e4
# golang.org/x/net/html/atom
runtime: failed to create new OS thread (have 2 already; errno=22)
fatal error: newosproc

runtime stack:
runtime.throw(0x612dd8, 0x9)
	/go/src/runtime/panic.go:596 +0x70
runtime.newosproc(0x10c24000, 0x10c34000)
	/go/src/runtime/os_linux.go:163 +0x158
runtime.newm(0x62d648, 0x0)
	/go/src/runtime/proc.go:1628 +0x10c
runtime.main.func1()
	/go/src/runtime/proc.go:126 +0x24
runtime.systemstack(0x8b6800)
	/go/src/runtime/asm_arm.s:264 +0x80
runtime.mstart()
	/go/src/runtime/proc.go:1132

goroutine 1 [running]:
runtime.systemstack_switch()
	/go/src/runtime/asm_arm.s:209 +0x4 fp=0x10c207bc sp=0x10c207b8
runtime.main()
	/go/src/runtime/proc.go:127 +0x58 fp=0x10c207e4 sp=0x10c207bc
runtime.goexit()
	/go/src/runtime/asm_arm.s:1017 +0x4 fp=0x10c207e4 sp=0x10c207e4
# github.com/mattn/go-runewidth
runtime: failed to create new OS thread (have 2 already; errno=22)
fatal error: newosproc

runtime stack:
runtime.throw(0x612dd8, 0x9)
	/go/src/runtime/panic.go:596 +0x70
runtime.newosproc(0x10c24000, 0x10c34000)
	/go/src/runtime/os_linux.go:163 +0x158
runtime.newm(0x62d648, 0x0)
	/go/src/runtime/proc.go:1628 +0x10c
runtime.main.func1()
	/go/src/runtime/proc.go:126 +0x24
runtime.systemstack(0x8b6800)
	/go/src/runtime/asm_arm.s:264 +0x80
runtime.mstart()
	/go/src/runtime/proc.go:1132

goroutine 1 [running]:
runtime.systemstack_switch()
	/go/src/runtime/asm_arm.s:209 +0x4 fp=0x10c207bc sp=0x10c207b8
runtime.main()
	/go/src/runtime/proc.go:127 +0x58 fp=0x10c207e4 sp=0x10c207bc
runtime.goexit()
	/go/src/runtime/asm_arm.s:1017 +0x4 fp=0x10c207e4 sp=0x10c207e4
# golang.org/x/net/proxy
runtime: failed to create new OS thread (have 2 already; errno=22)
fatal error: newosproc

runtime stack:
runtime.throw(0x612dd8, 0x9)
	/go/src/runtime/panic.go:596 +0x70
runtime.newosproc(0x10c26000, 0x10c36000)
	/go/src/runtime/os_linux.go:163 +0x158
runtime.newm(0x62d648, 0x0)
	/go/src/runtime/proc.go:1628 +0x10c
runtime.main.func1()
	/go/src/runtime/proc.go:126 +0x24
runtime.systemstack(0x8b6800)
	/go/src/runtime/asm_arm.s:264 +0x80
runtime.mstart()
	/go/src/runtime/proc.go:1132

goroutine 1 [running]:
runtime.systemstack_switch()
	/go/src/runtime/asm_arm.s:209 +0x4 fp=0x10c227bc sp=0x10c227b8
runtime.main()
	/go/src/runtime/proc.go:127 +0x58 fp=0x10c227e4 sp=0x10c227bc
runtime.goexit()
	/go/src/runtime/asm_arm.s:1017 +0x4 fp=0x10c227e4 sp=0x10c227e4
```

bjet007, 2017-08-30:
It seem that rebuilding golang with the golang-arm project (https://github.com/resin-io/golang-arm) fix the issue as it contains the golang patch.

nghiant2710, 2017-10-09:
@ensonik, I believe this patch https://github.com/resin-io/golang-arm/commit/df155921c8dbd517700723bf2e8d667bcbbde6ce with solve this issue. We will rebuild the Go binary soon to apply this patch. In the meantime, you can try out Go 1.9 base image which doesn't have this issue.

ensonik, 2017-10-09:
Thanks @nghiant2710 . Using the 1.9 image worked.

closed 2018-01-02 by nghiant2710