balena-io-library/base-images #343: Problem with python3.6 build: unsupported hash type blake2s

reinhrst, 2017-09-23:
Not sure if it's the build's fault or somewhere upstream:

```
$ docker run -ti --entrypoint python resin/qemux86-alpine-python:3.6                                                                      
Python 3.6.1 (default, Apr  5 2017, 15:43:26)
[GCC 6.2.1 20160822] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import hashlib
ERROR:root:code for hash blake2b was not found.
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/hashlib.py", line 243, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/usr/local/lib/python3.6/hashlib.py", line 119, in __get_openssl_constructor
    return __get_builtin_constructor(name)
  File "/usr/local/lib/python3.6/hashlib.py", line 113, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type blake2b
ERROR:root:code for hash blake2s was not found.
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/hashlib.py", line 243, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/usr/local/lib/python3.6/hashlib.py", line 119, in __get_openssl_constructor
    return __get_builtin_constructor(name)
  File "/usr/local/lib/python3.6/hashlib.py", line 113, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type blake2s
```

lurch, 2018-01-30:
@nghiant2710 Looks like https://stackoverflow.com/questions/20399331/error-importing-hashlib-with-python-2-7-but-not-with-2-6 has some info that'll be useful for debugging this problem

nghiant2710, 2018-02-27:
@reinhrst, I think it's somewhere upstream since the latest `2.7` and `3.6` images no longer have this issue. Please let us know if you need further support.

closed 2018-02-27 by nghiant2710