balena-io-library/base-images #387: Create a base image with OpenCV

LucianBuzzo, 2018-02-19:


sgtwilko, 2019-01-28:
I have been publishing OpenCV images for a little while based off the resin Raspbian images.
Feel free to grab my docker files and modify (i.e., clean up and make them pro grade!) to have a base image.

See here:
[SgtWilko/rpi-raspbian-opencv](https://hub.docker.com/r/sgtwilko/rpi-raspbian-opencv)

JeffJassky, 2019-08-12:
@LucianBuzzo I would also love to see this. Compiling OpenCV takes about 8 hours.

@sgtwilko Thanks for sharing this! I don't suppose you have any builds on Buster yet?

sgtwilko, 2019-08-12:
Hi Jeff,No I find the OpenCv compilation process fraught with issues where the RPi will freeze.This is more frequent with the balena images than the resin images (no idea why!)As such I'm a bit behind on updating to Buster.I have acquired a RPi 4 4GB which should speed things up.Keep an eye on the docket hub.SgtWilko.

sgtwilko, 2019-09-07:
@JeffJassky  I've pushed a Buster image to a test docker hub repo:
https://hub.docker.com/r/sgtwilko/balenalib-rpi-raspbian-opencv

I've not had a chance to test it yet, if you do get a chance, can you let me know if it works.

Thanks