balena-io-library/base-images #424: Tag latest and stretch not found.

4me2, 2018-05-23:
Since a few days a "docker pull resin/rpi-raspbian" the image failed with the message:
Pulling repository docker.io/resin/rpi-raspbian
Tag **_latest_** not found in repository docker.io/resin/rpi-raspbian

also tried :stretch but failed as well.
Pulling repository docker.io/resin/rpi-raspbian
Tag **_stretch_** not found in repository docker.io/resin/rpi-raspbian

Can you please help me out?

Thanks.

4me2, 2018-05-23:
@nghiant2710
Thanks for testing, it is possible someting is wrong on my side but pulling other images is no issue and nothing has been changed on this since the last succesful pull.

docker pull centos
Using default tag: latest
latest: Pulling from library/centos
25e42c0f2877: Pull complete 
6e8580b4bb45: Pull complete 
7b1ce1970405: Pull complete 
Digest: sha256:4b63b3e9fe127c07a15458af7fcc187980a9fff8069063057bbc9c282f26f539
Status: Downloaded newer image for centos:latest

docker pull ubuntu
Using default tag: latest
latest: Pulling from library/ubuntu
92ad863152ec: Pull complete 
48d65e8607c6: Pull complete 
d48c45472535: Pull complete 
ed5a16b55bcf: Pull complete 
d123588e5cab: Pull complete 
266050bcf8f9: Pull complete 
Digest: sha256:667c4b3820ddfbfe7b2dc9816123c936f45af6f31bd51b01d02e97dc5f09e3d5
Status: Downloaded newer image for ubuntu:latest

And finally:
docker pull hypriot/rpi-gogs-raspbian
Using default tag: latest
latest: Pulling from hypriot/rpi-gogs-raspbian
0b87ad1e8812: Pulling fs layer 
96aedd7daa45: Pulling fs layer 
efcbb14132c5: Pulling fs layer 
e93d329196f6: Pulling fs layer 
c52ee79bea7a: Pulling fs layer 
4ce00c22988c: Pulling fs layer 
0d739dd9b9aa: Pulling fs layer 
ff9fe6680de7: Pulling fs layer 
93589dbbb031: Pulling fs layer 
fb9830de48c3: Pulling fs layer 
ceebc9f096ac: Pulling fs layer 
e6e4676af372: Pulling fs layer 
3a131b4a41ae: Pulling fs layer 
872219db4528: Pulling fs layer 
ac985b3f2742: Pulling fs layer 
7e045dba3610: Download complete 
fb413215bea0: Pulling fs layer 
22fea935e8ed: Pulling fs layer 
94b636c32377: Pulling fs layer 
Pulling repository docker.io/hypriot/rpi-gogs-raspbian
4f3859bdccea: Download complete 
50cd794ffde5: Download complete 
d7b5badc2683: Download complete 
e528f9cdab20: Download complete 
eac7f4e88488: Download complete 
95cd88bb3fae: Download complete 
f932a331515c: Download complete 
2443f7d26316: Download complete 
1b8554f21204: Download complete 
f9c19d29e26a: Download complete 
a86d8434d760: Download complete 
db0e892f4e38: Download complete 
e93435e47a5f: Download complete 
f265db24d376: Download complete 
4ee6a0bcfdec: Download complete 
4f8ae367a944: Download complete 
85624dff9796: Download complete 
38c0a412433a: Download complete 
e5fce3220f3e: Download complete 
Status: Downloaded newer image for hypriot/rpi-gogs-raspbian:latest
docker.io/hypriot/rpi-gogs-raspbian: this image was pulled from a legacy registry.  Important: This registry version will not be supported in future versions of docker.

docker images
REPOSITORY                  TAG                 IMAGE ID            CREATED             VIRTUAL SIZE
ubuntu                      latest              266050bcf8f9        3 weeks ago         79.62 MB
centos                      latest              7b1ce1970405        6 weeks ago         198.6 MB
hypriot/rpi-gogs-raspbian   latest              4f3859bdccea        2 years ago         237.3 MB

nghiant2710, 2018-05-25:
@4me2 what is the docker version on your PI? As I recall, docker dropped support for old client and the docker version on your Pi maybe out of date.

4me2, 2018-05-25:
@nghiant2710 
docker --version
Docker version 1.8.3, build f4bf5c7

4me2, 2018-05-25:
@nghiant2710 

Ok, issue solved now:
Installed docker not from the Raspbian Stretch repository but used the following steps which I have found here: https://blog.alexellis.io/getting-started-with-docker-on-raspberry-pi/

curl -sSL https://get.docker.com | sh
sudo systemctl enable docker
sudo systemctl start docker
sudo usermod -aG docker pi
docker --version
Docker version 18.05.0-ce, build f150324

docker pull resin/rpi-raspbian
Using default tag: latest
latest: Pulling from resin/rpi-raspbian
e485920b0f83: Pull complete
61bc94ceadb2: Pull complete
7e88c0f82bf1: Pull complete
8427cc4400a5: Pull complete
9fb0e864fa2e: Pull complete
bbcfbd5da592: Pull complete
c689f1a0503f: Pull complete
b2cc8693f958: Pull complete
71cbe1fdb024: Pull complete
da0650675c82: Pull complete
bd1688cba618: Pull complete
Digest: sha256:4b874601f174cde3147da92417664d138c251fec74e083c6f667fec5cfa0cc8c
Status: Downloaded newer image for resin/rpi-raspbian:latest

@nghiant2710 
Thanks for your time and pointing me out.

closed 2018-05-25 by 4me2