balena-io-library/base-images #429: Support Ubuntu 16.04 for TX2

kgreenek, 2018-06-10:
The default distro installed on the TX2 is Ubuntu 16.04. We currently build everything for Ubuntu 16.04 and are exploring migrating to a docker-based approach using resin. One of the major pain points that we've been struggling with is version incompatibilities between debian's package versions and Ubuntu's equivalent packages.

The biggest problem has been gcc versions. Ubuntu 16.04 uses gcc 5.4, but there is no way to install gcc 5.4 on any version of debian (short of building gcc from source). There are also some difficulties with cuda, opencv, etc. All of the TX2 tooling is built around Ubuntu 16.04.

Supporting a docker image based on Ubuntu 16.04 would make migrating to resin more-or-less seamless for us on the TX2.

nghiant2710, 2018-06-14:
@kgreenek, I'm working on adding ubuntu to our base images so it will be available soon. We will let you know once when the ubuntu base images are ready.

joshughes, 2018-06-18:
@kgreenek we had the same issue... if your blocked feel free to use the base image I created until this PR is merged 

jjhughes57/jetson-tx2-ubuntu:xenial3

It has been working without any issue for our TX2s

nghiant2710, 2018-09-12:
Ubuntu base images were released https://github.com/resin-io-library/base-images/pull/431!

closed 2018-09-12 by nghiant2710