balena-io-library/base-images #433: Missing /sbin/init in Debian Stretch for Raspberry PI

splitice, 2018-06-20:
Issue #319 was never fixed on the rpi images.

nghiant2710, 2018-06-20:
@splitice, it was fixed in https://github.com/resin-io-library/resin-rpi-raspbian/pull/76. I just tested the image, `/sbin/init` is there and you can enable systemd properly.

splitice, 2018-06-20:
Perhaps check the Raspberry PI images, not Raspbian - those images are different. In ```resin/raspberry-pi2-debian:stretch``` there is no /sbin/init.

nghiant2710, 2018-06-20:
I just pulled `resin/raspberry-pi2-debian:stretch` to check locally and pushed that image to a device and have no problem with it.
```
docker run -ti --rm resin/raspberry-pi2-debian:stretch bash
[WARN  tini (24)] Tini is not running as PID 1 and isn't registered as a child subreaper.
Zombie processes will not be re-parented to Tini, so zombie reaping won't work.
To fix the problem, use the -s option or set the environment variable TINI_SUBREAPER to register Tini as a child subreaper, or run Tini as PID 1.
root@e870887f6acf:/# cd /sbin
root@e870887f6acf:/sbin# ls
agetty         findfs         ip                modprobe      start-stop-daemon
badblocks      fsck           ipmaddr           nameif        sulogin
blkdeactivate  fsck.cramfs    iptunnel          pam_tally     swaplabel
blkdiscard     fsck.ext2      isosize           pam_tally2    swapoff
blkid          fsck.ext3      killall5          pivot_root    swapon
blockdev       fsck.ext4      ldconfig          plipconfig    switch_root
bridge         fsck.minix     logsave           poweroff      sysctl
cfdisk         fsfreeze       losetup           rarp          tc
chcpu          fstab-decode   lsmod             raw           telinit
ctrlaltdel     fstrim         mii-tool          reboot        tini
debugfs        getty          mke2fs            resize2fs     tipc
depmod         halt           mkfs              rmmod         tune2fs
devlink        hwclock        mkfs.bfs          route         udevadm
dmsetup        ifconfig       mkfs.cramfs       rtacct        unix_chkpwd
dmstats        ifdown         mkfs.ext2         rtmon         unix_update
dumpe2fs       ifquery        mkfs.ext3         runlevel      wipefs
e2fsck         ifup           mkfs.ext4         runuser       zramctl
e2image        init           mkfs.minix        sfdisk
e2label        initctl        mkhomedir_helper  shadowconfig
e2undo         insmod         mkswap            shutdown
fdisk          installkernel  modinfo           slattach
```

You can find `init` there.
By any chances, you're using an old image? Can you remove and pull it again?

splitice, 2018-06-20:
I can't have an old image, the build is done by the resin build servers.

nghiant2710, 2018-06-20:
@splitice can you please share your Dockerfile with me please?

splitice, 2018-06-20:
I can not. However the only part that should matter.

```
# pull base image
FROM resin/%%RESIN_MACHINE_NAME%%-debian:stretch
```

I can confirm that rpi2 jessie works fine, as does the qemu-x64 stretch.

nghiant2710, 2018-06-20:
I just pushed this `Dockerfile.template` to a rpi2 application
```
FROM resin/%%RESIN_MACHINE_NAME%%-debian:stretch

ENV INITSYSTEM on

RUN apt-get update && apt-get install -y python

COPY src/app.py .

CMD ["python", "app.py"]
```

and here is the first part of my build log which indicates that `resin/raspberry-pi2-debian:stretch` was used to build the image
```
$ git push testrpi2 master
Counting objects: 2038, done.
Delta compression using up to 8 threads.
Compressing objects: 100% (2022/2022), done.
Writing objects: 100% (2038/2038), 1.77 MiB | 2.76 MiB/s, done.
Total 2038 (delta 1135), reused 0 (delta 0)

[Info]  Starting build for testrpi2, user nghiant2710
[Info]  Dashboard link: https://dashboard.resin.io/apps/1133367/devices
[Info]  Building on arm03
[Info]  Pulling previous images for caching purposes...
[Success]  Successfully pulled cache images
[main]  Step 1/5 : FROM resin/raspberry-pi2-debian:stretch
[main]   ---> 9dbf01ddb52f
[main]  Step 2/5 : ENV INITSYSTEM on
[main]   ---> Running in f29f0cbf18a7
[main]   ---> 86512dc273fc
[main]  Removing intermediate container f29f0cbf18a7
[main]  Step 3/5 : RUN apt-get update && apt-get install -y python
[main]   ---> Running in 339a7eaf18d1
[main]  Get:1 http://security.debian.org/debian-security stretch/updates InRelease [94.3 kB]
```
and I have no issue :/ Can you please share with us a part of the build log or error message when the container starts so I can reproduce your issue? Is there any chance, your build was cached?

splitice, 2018-06-20:
The error you will see is:

```
20.06.18 15:41:51 (+1000)  main  Systemd init system enabled.
20.06.18 15:41:52 (+1000)  main  env: ‘/sbin/init’: No such file or directory
```

I modified some files to get into the container and was able to see that indeed there is no /sbin/init, no systemd package installed at all.

The build with ```apt-get install systemd-sysv``` in the docker file worked fine.

As this is a build conducted on the ResinOS build system, I have no control over any caching.

nghiant2710, 2018-06-20:
Indeed, `systemd-sysv` was missing and we installed it in the PR to fix the issue but I don't know why you're having this issue since it's fixed a while ago and the images on dockerhub have the fix applied.
Are you using `resin-cli` to push to your device?

splitice, 2018-06-20:
Nope, just a git push and device upgrade (both delta and non-delta tested)

nghiant2710, 2018-06-20:
Can you please try pushing to `resin-nocache` branch which will rebuild your Dockerfile without using any caches, ie `git push resin <originbranch>:resin-nocache` since owe use Docker layer caching on the build servers to speed up builds

splitice, 2018-06-20:
Wouldn't the bottom layers be invalidated though when they get updated?

I'll remove the apt-get and try that out tomorrow. For now though that apt-get install of systemd is certainly working.

nghiant2710, 2018-06-20:
@splitice, we pull the cache first so if you don't change much in your Dockerfile then the bottom layers will be pulled from the cache images. Anyway, please let us know how it goes.

splitice, 2018-06-21:
Yes there seems to have been some type of caching as a fresh app doesnt show the same result. This app is always pushing to nocache as it relies upon delta updates and most of our changing code is external docker file.

Not going to be a big issue for us now that we can apt-get the package in as a workaround.

closed 2018-06-22 by nghiant2710