balena-io-library/base-images #439: Python stretch images

FRVisser, 2018-07-02:
Hey, 

We run in some issues with the git version and this would be solved if we could use debian stretch images. I see stretch dockerfiles but there aren't any tags to use it. 

We would like to use a 3.6 slim stretch image. Is the release of stretch in the pipeline?

Tthanks!

FluxReality, 2018-09-21:
@nghiant2710 are there any updates on this?

Much in need for stretch images of python.

nghiant2710, 2018-09-22:
@FRVisser @FluxReality there is a `3.7-stretch` image that you can use now. We have a little issue with other versions due to a new OpenSSL version on Debian Stretch (we're compiling all the binaries on Debian Jessie and they won't work on Stretch) so support for other versions will be added later.

FRVisser, 2018-10-02:
I works like a dream. However, I have one more request: We used to use the 3-slim image because of its small size. The bare size difference is now 226 Mb for the old 3-slim image and 790 Mb for the new stretch image. 

Any change there will be stretch-slim image available in the future?

nghiant2710, 2019-01-28:
@FRVisser regarding your requests here, I guess you will love the new balenalib base images

FRVisser, 2019-01-28:
@nghiant2710 71 MB 😱 What secret lies beneath this beautiful number?

closed 2019-02-01 by nghiant2710