balena-io-library/base-images #444: Failing to build image based on intel-nuc-python

miccio-dk, 2018-07-17:
I'm trying to deploy a container locally using `sudo resin local push -s path/to/dockerfile/`.
However, the build process fails halfway:
````
Reporting discovered devices
? select a device 79932cf.local (10.0.111.111)
* Building..
- Stopping and removing any previous 'local-app' container
- Uploading build context & starting build...
Step 1/6 : FROM resin/intel-nuc-python:3.5
 ---> 996e253cd45a
Step 2/6 : WORKDIR /usr/src/app
 ---> 80f580e48419
Removing intermediate container daa90f5e337b
Step 3/6 : COPY ./requirements.txt /requirements.txt
Push failed. Error: failed to export image: failed to set parent sha256:80f580e48419fc38e1ec34411c9171c7da0bc53011fff8df3cdeab25f5fb1aeb: unknown parent image ID sha256:80f580e48419fc38e1ec34411c9171c7da0bc53011fff8df3cdeab25f5fb1aeb Error: failed to export image: failed to set parent sha256:80f580e48419fc38e1ec34411c9171c7da0bc53011fff8df3cdeab25f5fb1aeb: unknown parent image ID sha256:80f580e48419fc38e1ec34411c9171c7da0bc53011fff8df3cdeab25f5fb1aeb
    at Stream.<anonymous> (/home/riccardo/.npm-global/lib/node_modules/resin-cli/node_modules/resin-sync/build/docker-utils.js:146:23)
    at emitOne (events.js:116:13)
    at Stream.emit (events.js:211:7)
    at drain (/home/riccardo/.npm-global/lib/node_modules/resin-cli/node_modules/through/index.js:36:16)
    at Stream.stream.queue.stream.push (/home/riccardo/.npm-global/lib/node_modules/resin-cli/node_modules/through/index.js:45:5)
    at Parser.parser.onToken (/home/riccardo/.npm-global/lib/node_modules/resin-cli/node_modules/JSONStream/index.js:129:18)
    at Parser.proto.write (/home/riccardo/.npm-global/lib/node_modules/resin-cli/node_modules/jsonparse/jsonparse.js:135:34)
    at Stream.<anonymous> (/home/riccardo/.npm-global/lib/node_modules/resin-cli/node_modules/JSONStream/index.js:21:12)
    at Stream.stream.write (/home/riccardo/.npm-global/lib/node_modules/resin-cli/node_modules/through/index.js:26:11)
    at IncomingMessage.ondata (_stream_readable.js:639:20)
    at emitOne (events.js:116:13)
    at IncomingMessage.emit (events.js:211:7)
    at addChunk (_stream_readable.js:263:12)
    at readableAddChunk (_stream_readable.js:250:11)
    at IncomingMessage.Readable.push (_stream_readable.js:208:10)
    at HTTPParser.parserOnBody (_http_common.js:141:22)
    at Socket.socketOnData (_http_client.js:440:20)
    at emitOne (events.js:116:13)
    at Socket.emit (events.js:211:7)
    at addChunk (_stream_readable.js:263:12)
    at readableAddChunk (_stream_readable.js:250:11)
    at Socket.Readable.push (_stream_readable.js:208:10)
    at TCP.onread (net.js:594:20)
````
Note: the hashes mentioned in the messages are different everytime.

This is my Dockerfile:
````
FROM resin/intel-nuc-python:3.5
WORKDIR /usr/src/app
COPY ./requirements.txt /requirements.txt
RUN pip install -r /requirements.txt
COPY . ./
CMD ["python", "config_fetcher.py", "-n"]
`````

For completeness sake, this is my `requirements.txt`:
`````
requests
python-dotenv
``````

I tried using the `-onbuild` version and  removing the lines handling `requirements.txt` but it didn't help. 

Deleting the `.resin-sync.yml` and repeating the process yields this (slightly different) error message:
````Reporting discovered devices
? select a device 79932cf.local (10.0.111.111)
* Building..
- Stopping and removing any previous 'local-app' container
- Uploading build context & starting build...
Step 1/6 : FROM resin/intel-nuc-python:3.5
 ---> 996e253cd45a
Step 2/6 : WORKDIR /usr/src/app
 ---> 56139f9ca85d
Removing intermediate container 2a6fb003e853
Step 3/6 : COPY ./requirements.txt /requirements.txt
Push failed. Error: failed to get destination image "sha256:56139f9ca85dcb640229613667eec23144ba770ce8fb575624ec5f29afe97318": failed to get layer for image sha256:56139f9ca85dcb640229613667eec23144ba770ce8fb575624ec5f29afe97318: layer does not exist Error: failed to get destination image "sha256:56139f9ca85dcb640229613667eec23144ba770ce8fb575624ec5f29afe97318": failed to get layer for image sha256:56139f9ca85dcb640229613667eec23144ba770ce8fb575624ec5f29afe97318: layer does not exist
    at Stream.<anonymous> (/home/riccardo/.npm-global/lib/node_modules/resin-cli/node_modules/resin-sync/build/docker-utils.js:146:23)
    at emitOne (events.js:116:13)
    at Stream.emit (events.js:211:7)
    at drain (/home/riccardo/.npm-global/lib/node_modules/resin-cli/node_modules/through/index.js:36:16)
    at Stream.stream.queue.stream.push (/home/riccardo/.npm-global/lib/node_modules/resin-cli/node_modules/through/index.js:45:5)
    at Parser.parser.onToken (/home/riccardo/.npm-global/lib/node_modules/resin-cli/node_modules/JSONStream/index.js:129:18)
    at Parser.proto.write (/home/riccardo/.npm-global/lib/node_modules/resin-cli/node_modules/jsonparse/jsonparse.js:135:34)
    at Stream.<anonymous> (/home/riccardo/.npm-global/lib/node_modules/resin-cli/node_modules/JSONStream/index.js:21:12)
    at Stream.stream.write (/home/riccardo/.npm-global/lib/node_modules/resin-cli/node_modules/through/index.js:26:11)
    at IncomingMessage.ondata (_stream_readable.js:639:20)
    at emitOne (events.js:116:13)
    at IncomingMessage.emit (events.js:211:7)
    at addChunk (_stream_readable.js:263:12)
    at readableAddChunk (_stream_readable.js:250:11)
    at IncomingMessage.Readable.push (_stream_readable.js:208:10)
    at HTTPParser.parserOnBody (_http_common.js:141:22)
    at Socket.socketOnData (_http_client.js:440:20)
    at emitOne (events.js:116:13)
    at Socket.emit (events.js:211:7)
    at addChunk (_stream_readable.js:263:12)
    at readableAddChunk (_stream_readable.js:250:11)
    at Socket.Readable.push (_stream_readable.js:208:10)
    at TCP.onread (net.js:594:20)
````

Sometimes the following error would appear instead of the _failed to get destination image_
````
Push failed. Error: failed to get digest sha256:a18a67288b360c59f8af69facc888a37c6e230129a5772fd38695788ce4d65d1: open /var/lib/docker/image/aufs/imagedb/content/sha256/a18a67288b360c59f8af69facc888a37c6e230129a5772fd38695788ce4d65d1: no such file or directory Error: failed to get digest sha256:a18a67288b360c59f8af69facc888a37c6e230129a5772fd38695788ce4d65d1: open /var/lib/docker/image/aufs/imagedb/content/sha256/a18a67288b360c59f8af69facc888a37c6e230129a5772fd38695788ce4d65d1: no such file or directory
`````

Any idea on what could be wrong or missing? Thanks in advance.

shaunmulligan, 2018-07-17:
@miccio-dk can you ensure that you have set your device into "localMode" via the dashboard action, perhaps enable and re-enable it just to make sure. I have only ever seen this issue when using local push on a device that is not in "localMode"

CameronDiver, 2018-07-17:
@miccio-dk this looks to me like you're trying to use local mode without first enabling local mode for this device.

closed 2018-07-24 by nghiant2710