balena-io-library/base-images #453: Signals are not sent to processes in recent base images

imrehg, 2018-08-03:
In recent base images don't seem to be able to catch signals (such as `SIGINT`). Seems like the inability to catch the signals might have started around the change to `tini`. Was using this test project:

* Python test: https://github.com/resin-io-playground/signalstest/tree/master
* Shell test: https://github.com/resin-io-playground/signalstest/tree/debian

Deploying this project to a resin device, and stopping or restarting the container, so a SIGINT is sent.

In both cases, base images before the 2017/06/20 [switch to tini](https://github.com/resin-io-library/base-images/blob/master/CHANGELOG.md#2017-06-20) seem to work, while shortly after doesn't. Non-resin base images (Python or Debian, above) also seem to work with the same code.

References:
* [Forums](https://forums.resin.io/t/cant-capture-signals-from-simple-python-application/3561)
* [Flowdock](https://www.flowdock.com/app/rulemotion/i-baseimages/threads/N0Xh4VwA28QxLlibdD3zBTgpetP)

Krismix1, 2018-08-24:
Do you have any updates regarding the issue?

nghiant2710, 2018-08-24:
@Krismix1, I'm still investigating the issue, It seems like `Tini` is not the root cause here since I did the above test locally and it works fine all the times but the behavior is a bit different on resin device when the script sometimes won't catch the signal correctly.

We will keep investigating and will let you know when it's done.

Krismix1, 2018-09-20:
Hey guys, did you get any progress with it?

pcarranzav, 2018-09-28:
Connects to https://github.com/resin-io/hq/issues/1483

(Still investigating!)

johnslemmer, 2018-11-06:
I believe I'm running in to the same issue.  I'm developing a node application that utilizes [`pigpio`](https://github.com/fivdi/pigpio#readme).  `pigpio` has interrupt handler code to cleanup nicely on `SIGINT`, `SIGTERM`, etc. but those signals never come.  `pigpio` is really fragile to `SIGKILL` and actually requires a hard reboot of the raspberry-pi if the process isn't cleaned up correctly. This is a _really_ bothersome issue.  Let me know how I can be helpful. I can even put up another example repo that replicates this issue if that would help.

andrewn, 2019-01-18:
I'm having the same problem with `pigpio` (which is used by the [raspi-io/raspi-gpio](https://github.com/nebrius/raspi-io)). Deploying a new balena app doesn't work without a reboot because without receiving the signal, my app can't clean up properly.

Are there any workarounds to get these signals working? Or is there perhaps an older node.js image that I can use until this issue is resolved?

adipascu, 2019-05-19:
This issue is fixed in the latest images.

wlisac, 2019-05-24:
I'm seeing this fixed in `raspberrypi3` base images, but not fixed in `raspberry-pi` base images.

nghiant2710, 2019-05-24:
@wlisac, can you please elaborate? What image did you use?

This problem should be fixed in all `balenalib` base images

wlisac, 2019-05-24:
@nghiant2710 sorry for the confusion – I ran my tests again and the images are working as expected. Thanks!

closed 2019-05-27 by nghiant2710