balena-io-library/base-images #464: Raspberry Pi Rust base images

rahul-thakoor, 2018-09-09:
Hello there,
I currently use Rust on resinOS by adapting the hello-world example at https://github.com/resin-io-projects/resin-rust-hello-world/blob/master/Dockerfile.template

Is there a base image for Rust I can start with? So I could do 

> FROM resin/%%RESIN_MACHINE_NAME%%-rust

I would be interested to implement it if it doesn't exist and if some mentoring is provided. I can test on Raspberry Pi 1 B+, Raspberry Pi Zero W and Raspberry Pi 3B+

Thanks

FibreFoX, 2018-11-10:
I found this here, maybe this helps a bit? I'm just new to docker + raspberry pi so maybe this might not be what you searched for:
https://github.com/jimmycuadra/docker-rust-armv7hf

imrehg, 2018-11-11:
Hi, as much as I know there are some issues using rust in our cloud building environment that we are working through. This is an example project of how it can work, using multistage docker builds in the template.

https://github.com/balena-io-projects/balena-rust-hello-world

It's not 100% straightforward, but should be manageable. We'd love some rust base image, and working towards it, just not yet.