balena-io-library/base-images #465: Pigpio failing on lack of permissions using resin/raspberrypi3-node

mdbeauche, 2018-09-11:
I'm having issues getting the node package pigpio (which is a wrapper for [pigpio in C](https://github.com/joan2937/pigpio)) (I'm using the package to access GPIO pins) to run in a default Docker container running default resinOS resin/raspberrypi3-node ([GitHub issue here](https://github.com/joan2937/pigpio/issues/235), [Resin.io forum post here](https://forums.resin.io/t/pigpio-failing-on-lack-of-permissions-using-resin-raspberrypi3-node/3917)).

Dockerfile:

```
FROM resin/raspberrypi3-node

RUN apt-get update && apt-get install -yq --no-install-recommends pigpio

RUN npm install pigpio
```

From running docker ( `docker run -it name /bin/bash`  and then  `sudo pigpiod`  at shell as root - pigpiod is the pigpio daemon):

```
2018-09-06 19:55:38 initCheckPermitted: 
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

root@1bd67dc5c6e4:/# Can't initialise pigpio library
```

I'm already running as root, and using sudo. I've tried changing docker run permissions ( `--privileged`  and  `--cap-add=ALL` and `--privileged -v /sys:/sys -v /dev/mem:/dev/mem` ) and I get the same result. Same error running the daemon as if I try to use the pigpio node package in a script.

carylwyatt, 2018-09-21:
I had a bunch of issues with `pigpio` when first building my dockerized app. I ended up adding `RUN apt-get install pigpio` to my Dockerfile after the `RUN apt-get update` line. I did try installing it via npm like you're doing here, but I could never get it to work. Apt-get did the trick. I also have a `apt-get clean` run statement before update, but I'm not really sure if that helps anything.

I also mount a volume, but mine is `-v /lib/modules:/lib/modules` ...but this might have more to do with the modprobe/i2c issues I was having. It's hard to remember.

curcuz, 2019-01-25:
Hi there, are you trying to deploy a multicontainer application (ie do you have a docker-compose.yml file in your repository root)? if so, can you please try setting the following label to your container?

```
privileged: true
```

this is ofc just to test if the issue is related the container not having priviledges to access GPIOs. Then, if that works, we will see what interface  `pigpio` uses and add the proper settings instead of the very broad `privileged: true`

carylwyatt, 2019-01-25:
Not using docker compose. 

It's been a while since I've worked on this project, but looking at my docs, I use the `--privileged` flag in my run statement, so, yes? 

I'd have to go back and read my notes/do some more testing, but I'm pretty sure this line from my Dockerfile is what fixed my pigpio issues:

`RUN apt-get install pigpio python-pigpio python3-pigpio`

I could never get this to work in an alpine node environment because I couldn't install `pigpio` via `apk` and the make file would never build properly for me (probably user error, I'm a javascript girl and know nothing about C).

Feel free to peruse my repo: https://github.com/carylwyatt/docker-people-counter