balena-io-library/base-images #467: Image size

zero-88, 2018-09-13:
Hi,
I'm curious what inside of container, it is quite heavy, at least for `alpine` image.
```bash
resin/raspberrypi3-alpine-node                           8                   9cc6566f47e9        39 hours ago        460MB
docker.io/node                          8-alpine            df48b68da02a        26 hours ago        65.7 MB
```

Just compare package list between them:
- `docker.io/node:8-alpine`
```bash
musl-1.1.19-r10 x86_64 {musl} (MIT) [installed]
libstdc++-6.4.0-r8 x86_64 {gcc} (GPL LGPL) [installed]
zlib-1.2.11-r1 x86_64 {zlib} (zlib) [installed]
apk-tools-2.10.1-r0 x86_64 {apk-tools} (GPL2) [installed]
musl-utils-1.1.19-r10 x86_64 {musl} (MIT BSD GPL2+) [installed]
alpine-baselayout-3.1.0-r0 x86_64 {alpine-baselayout} (GPL-2.0) [installed]
libgcc-6.4.0-r8 x86_64 {gcc} (GPL LGPL) [installed]
alpine-keys-2.1-r1 x86_64 {alpine-keys} (MIT) [installed]
busybox-1.28.4-r1 x86_64 {busybox} (GPL-2.0) [installed]
libressl2.7-libcrypto-2.7.4-r0 x86_64 {libressl} (custom) [installed]
scanelf-1.2.3-r0 x86_64 {pax-utils} (GPL-2.0) [installed]
libressl2.7-libtls-2.7.4-r0 x86_64 {libressl} (custom) [installed]
libc-utils-0.7.1-r0 x86_64 {libc-dev} (BSD) [installed]
ssl_client-1.28.4-r1 x86_64 {busybox} (GPL-2.0) [installed]
libressl2.7-libssl-2.7.4-r0 x86_64 {libressl} (custom) [installed]
```

- `resin/raspberrypi3-alpine-node:8`
```bash
usbutils-009-r0 armhf {usbutils} (GPL-2.0-or-later) [installed]
pkgconf-1.5.3-r0 armhf {pkgconf} (ISC) [installed]
subversion-1.10.0-r0 armhf {subversion} (Apache-2.0 BSD) [installed]
libfdisk-2.32-r0 armhf {util-linux} (GPL-2.0 GPL-2.0-or-later LGPL-2.0-or-later BSD Public-Domain) [installed]
wget-1.19.5-r0 armhf {wget} (GPL-3.0-or-later) [installed]
g++-6.4.0-r8 armhf {gcc} (GPL LGPL) [installed]
bzip2-dev-1.0.6-r6 armhf {bzip2} (BSD) [installed]
libxext-1.3.3-r2 armhf {libxext} (custom) [installed]
libxau-1.0.8-r2 armhf {libxau} (custom) [installed]
libusb-1.0.22-r0 armhf {libusb} (LGPL-2.0-or-later) [installed]
less-530-r0 armhf {less} (GPL) [installed]
lz4-libs-1.8.2-r0 armhf {lz4} (BSD-2-Clause GPL-2.0-only) [installed]
sqlite-libs-3.24.0-r0 armhf {sqlite} (Public-Domain) [installed]
autoconf-2.69-r2 armhf {autoconf} (GPL-2.0-or-later) [installed]
libc-dev-0.7.1-r0 armhf {libc-dev} (BSD) [installed]
sqlite-dev-3.24.0-r0 armhf {sqlite} (Public-Domain) [installed]
pango-1.40.14-r0 armhf {pango} (LGPL) [installed]
mpfr3-3.1.5-r1 armhf {mpfr3} (GPL LGPL) [installed]
libcrypto1.0-1.0.2o-r2 armhf {openssl} (openssl) [installed]
jpeg-dev-8-r6 armhf {jpeg} (AS-IS) [installed]
libuv-1.20.2-r0 armhf {libuv} (MIT BSD ISC) [installed]
libx11-1.6.5-r1 armhf {libx11} (custom:XFREE86) [installed]
librsvg-2.40.20-r0 armhf {librsvg} (LGPL-2.0+) [installed]
glib-dev-2.56.1-r0 armhf {glib} (GPL) [installed]
nghttp2-libs-1.32.0-r0 armhf {nghttp2} (MIT) [installed]
udev-init-scripts-32-r2 armhf {udev-init-scripts} (GPL-2.0-only) [installed]
bzr-2.7.0-r1 armhf {bzr} (GPL-2.0-or-later) [installed]
musl-1.1.19-r10 armhf {musl} (MIT) [installed]
serf-1.3.9-r4 armhf {serf} (Apache-2.0) [installed]
libksba-1.3.5-r0 armhf {libksba} (GPL) [installed]
dbus-1.10.24-r1 armhf {dbus} (AFL-2.1 OR GPL-2.0+) [installed]
libbz2-1.0.6-r6 armhf {bzip2} (BSD) [installed]
bash-4.4.19-r1 armhf {bash} (GPL-3.0-or-later) [installed]
xz-5.2.4-r0 armhf {xz} (GPL-2.0-or-later Public-Domain) [installed]
isl-0.18-r0 armhf {isl} (MIT) [installed]
libcurl-7.61.1-r0 armhf {curl} (MIT) [installed]
db-5.3.28-r0 armhf {db} (custom) [installed]
libldap-2.4.46-r0 armhf {openldap} (custom) [installed]
git-2.18.0-r0 armhf {git} (GPL-2.0-or-later) [installed]
gdbm-1.13-r1 armhf {gdbm} (GPL) [installed]
pcre-8.42-r0 armhf {pcre} (BSD-3-Clause) [installed]
[... 114 lines cut ...]
libssl1.0-1.0.2o-r2 armhf {openssl} (openssl) [installed]
expat-2.2.5-r0 armhf {expat} (MIT) [installed]
libpcre32-8.42-r0 armhf {pcre} (BSD-3-Clause) [installed]
hwdata-usb-0.309-r0 armhf {hwdata} (GPL-2.0-or-later) [installed]
pcre-dev-8.42-r0 armhf {pcre} (BSD-3-Clause) [installed]
libxslt-dev-1.1.32-r0 armhf {libxslt} (custom) [installed]
libxrender-0.9.10-r2 armhf {libxrender} (custom) [installed]
imagemagick-libs-7.0.7.32-r0 armhf {imagemagick} (ImageMagick) [installed]
cups-libs-2.2.6-r0 armhf {cups} (GPL LGPL-2.0-or-later) [installed]
libpcrecpp-8.42-r0 armhf {pcre} (BSD-3-Clause) [installed]
dbus-libs-1.10.24-r1 armhf {dbus} (AFL-2.1 OR GPL-2.0+) [installed]
libcap-2.25-r1 armhf {libcap} (GPL) [installed]
imagemagick-7.0.7.32-r0 armhf {imagemagick} (ImageMagick) [installed]
gettext-dev-0.19.8.1-r2 armhf {gettext} (GPL-3.0+ AND LGPL-2.1+ AND MIT) [installed]
findutils-4.6.0-r1 armhf {findutils} (GPL-3.0+) [installed]
tiff-4.0.9-r6 armhf {tiff} (libtiff) [installed]
openrc-0.35.5-r4 armhf {openrc} (BSD-2-Clause) [installed]
pcre2-10.31-r0 armhf {pcre2} (BSD-3-Clause) [installed]
libressl2.7-libssl-2.7.4-r0 armhf {libressl} (custom) [installed]
pixman-0.34.0-r5 armhf {pixman} (custom) [installed]
```
Are these packages necessary? Could you point me out which one is specified for `raspberrypi3`?
I'm new on IoT domain, but I'm sure docker image is designed for lightweight, especially for IoT device with restrict storage.

Best,

nghiant2710, 2018-09-14:
@zero-88, the Resin base images are a bit different from official Docker images, you can find more details in our docs here https://docs.resin.io/reference/base-images/base-images/#resin-io-base-images.

Basically, the node images you're mentioning here is a combination of `rapsberrypi3` specific base image, `buildpack-deps` and `node`. The `raspberrypi3` image contains some specific packages for RPI (for example: https://github.com/resin-io-library/base-images/blob/master/resin-base-images/device-base/raspberrypi3/alpine/3.8/Dockerfile), while the `buildpack-deps` contains a wide range of common libs so this layer will take the major share of image size and the `node` layer is a specific node version (npm and yarn included). That's why there is a huge different between the `full` variant of this node image with official one.

We do provide `slim` variant for this image which contains only the minimal packages needed to run node. If you check the `resin/raspberrypi3-alpine-node:8-slim` is only about `103MB`.

nghiant2710, 2018-09-18:
@zero-88, we respect the official Docker design, also this will benefit most of the users, I mean beginners or new users who just start using Docker. We do provide other variants like `slim`, `onbuild` so you can customize it on your own way when you know what you're doing.

zero-88, 2018-09-18:
`Resin` design is a slightly different with official Docker design. When play `resin` in first time, it make confusing for me.
Docker official has: `<version>`, `<version>-slim` (almost based on `Ubuntu`) and `<version>-alpine`.
`Resin` way: `ubuntu:<version>` == `ubuntu:<version>-onbuild`, `ubuntu:<version>-slim` | `alpine-version` == `alpine:<version>-onbuild`, `alpine:<version>-slim`

`Alpine` is almost make sense if it's lightweight, that's why I suggestion renaming as above.
It is just my recommendation, so you can decide a way that you're. 
Be able to close it now.
Thanks :)

nghiant2710, 2018-09-18:
@zero-88, thanks for the suggestion, we will take it into consideration for coming changes.

Also please don't hesitate to let us know if you have any questions or issues.

closed 2018-09-18 by nghiant2710