balena-io-library/base-images #477: entry.sh double-escapes ENV vars

wrboyce, 2018-10-17:
`printf '%q="%q"\n' "$var" "${!var}"`

both `%q` and quotes are unnecessary and result in double-escaping.

spiderkeys, 2018-12-07:
@wrboyce The change in your merged PR has resulted in the values in /etc/docker.env becoming no longer double quoted.

This has gone back and forth over several months between having the /etc/docker.env values being not double quoted, then double quoted, and now not again, causing unexpected failures in our software thrice. If the variables are not double quoted, then the env file cannot be sourced, so the double quotes were a welcome change. Can we please stabilize the decision around this?

nghiant2710, 2019-01-14:
hey @spiderkeys, sorry for the late response, it seems the docker.env like how it is now (the one in the latest image like `resin/armv7hf-debian:jessie`) can be sourced properly. Can you please explain how is your script broken?

closed 2019-05-24 by nghiant2710