balena-io-library/base-images #489: Add docker image for node v10 (current LTS)

feluelle, 2018-11-04:


nghiant2710, 2018-11-08:
@feluelle you can try `resin/raspberry-pi-node` image, it currently points to node `v.10.9.0`

Since we're busy with moving `resin` base images to new namespace `balenalib` with a lot of changes so our node images are not up to date with latest node versions, sorry for the inconvenience.

feluelle, 2018-11-08:
Thank you @nghiant2710. Is it possible to use the v10 tag? Because on the docker hub for `resin/raspberry-pi-node` I can not find a tag for any v10 release. I do not want to use `latest` or whatever it is currently pointing to, actually :/

nghiant2710, 2018-11-08:
@feluelle  Of course, there is a `resin/raspberry-pi-node:10` image (it's currently latest), docker hub doesn't show all the tags so you can't find `v10` there. It's better to check docker store https://store.docker.com/community/images/resin/raspberry-pi-node/tags

feluelle, 2018-11-08:
That works, thank you.

closed 2018-11-08 by feluelle