balena-io-library/base-images #504: ONBUILD steps make it impossible to use images in Linux based CI pipeline

mateodelnorte, 2018-12-04:
Hi there,

We're using balena images in a local CI pipeline. We have multi-stage build processes for building alpine containers. Unfortunately, we're not able to pull in containers defined in this repo because they specify ONBUILD commands – which blow up any inheriting container builds. 

Can the ONBUILD commands be removed from containers? ONBUILD is deprecated from what I understand, so I don't think this should be too radical.

ironsalsa, 2018-12-04:
I second this - that's a huge breaking change, as there's no way to overwrite that command and it prevents a "cross-build-start" from ever working.

ironsalsa, 2018-12-04:
@mateodelnorte my current workaround is using the old resin Docker hub images - they're a bit out of date, but at least our pipelines aren't down:

resin/{same naming conventions}

So balenalib/raspberrypi3-alpine can be swapped out for resin/raspberrypi3-alpine as long as the version tag you're using is available.

mateodelnorte, 2018-12-04:
I'm just copy/pasting the contents of these images and creating my own. Working so far, but defeats the purpose of these images existing.

ironsalsa, 2018-12-04:
@pcarranzav this one may need help sooner than later - it's going to break CI and local builds.

nghiant2710, 2018-12-04:
@mateodelnorte, can you please share with us an example to reproduce your issue please?

mateodelnorte, 2018-12-04:
Sure thing: 

Dockerfile:
```
FROM balenalib/artik530-alpine-node:3.8-build
```
When run on a linux system:
```
docker build -f Dockerfile .
Sending build context to Docker daemon   16.9kB
Step 1/1 : FROM balenalib/artik530-alpine-node:3.8-build
3.8-build: Pulling from balenalib/artik530-alpine-node
905674ea9d94: Already exists
d91fe322e169: Already exists
a08d25c15fb7: Already exists
0ee0aa6bf225: Already exists
8e2a90d9330a: Already exists
8a7de0acae5d: Already exists
7e6e613851e6: Already exists
23d907e3ebd8: Already exists
e08f0d998d20: Already exists
47de92860c23: Already exists
06cb7f1dd963: Already exists
5b644253612a: Already exists
995c9d2ba50b: Already exists
Digest: sha256:e487096da13697f9d133a2173d2ecc5fe48242fc380867090fa77d84a0367223
Status: Downloaded newer image for balenalib/artik530-alpine-node:3.8-build
# Executing 1 build trigger
 ---> Running in 2248ce5cd490
standard_init_linux.go:190: exec user process caused "exec format error"
The command '/bin/sh -c cat /.balena/image-info' returned a non-zero code: 1
```

petrosagg, 2018-12-04:
@mateodelnorte thanks for raising this issue. We completely missed that detail of using cross-build-start in combination with ONBUILD.

We're discussing internally on how to fix this and will release new images asap.

mateodelnorte, 2018-12-04:
Thanks! Happy to help hammer out issues and make all these images a great benefit for the community.

closed 2019-01-28 by nghiant2710