balena-io-library/base-images #513: Add Zulu as another version for OpenJDK stack

nghiant2710, 2019-01-22:


emilm, 2019-05-20:
I successfully did this based on raspberry-pi-alpine:3.9

```docker
FROM balenalib/raspberry-pi-alpine:3.9

# http://cdn.azul.com/zulu-embedded/bin/zulu8.36.0.152-ca-jdk1.8.0_202-linux_aarch32hf.tar.gz
RUN mkdir -p /opt/workspace
WORKDIR /opt/workspace

# Java Version and other ENV
ENV JAVA_HOME=/usr/lib/jvm/zulu-8 \
    PATH=${PATH}:${JAVA_HOME}/bin \
    GLIBC_VERSION_MAJOR=2.22 \
    GLIBC_VERSION=2.22-r8 \
    LANG=C.UTF-8

RUN apk -U upgrade && \
    apk add --no-cache jattach --repository http://dl-cdn.alpinelinux.org/alpine/edge/community/ && \
    apk add libstdc++ curl ca-certificates bash java-cacerts tzdata && \
    for pkg in glibc-${GLIBC_VERSION} glibc-bin-${GLIBC_VERSION} glibc-i18n-${GLIBC_VERSION}; do curl -sSL https://github.com/armhf-docker-library/alpine-pkg-glibc/releases/download/${GLIBC_VERSION_MAJOR}/${pkg}.apk -o /tmp/${pkg}.apk; done && \
    apk add --allow-untrusted /tmp/*.apk && \
    rm -v /tmp/*.apk && \
    ( /usr/glibc-compat/bin/localedef --force --inputfile POSIX --charmap UTF-8 C.UTF-8 || true ) && \
    echo "export LANG=C.UTF-8" > /etc/profile.d/locale.sh && \
    /usr/glibc-compat/sbin/ldconfig /lib /usr/glibc-compat/lib

RUN ZULU_ARCH=zulu8.36.0.152-ca-jdk1.8.0_202-linux_aarch32hf.tar.gz && \
    INSTALL_DIR=$( dirname $JAVA_HOME ) && \
    ZULU_DIR=$( basename ${ZULU_ARCH} .tar.gz ) && \
    wget -q https://cdn.azul.com/zulu-embedded/bin/${ZULU_ARCH} && \
    mkdir -p ${INSTALL_DIR} && \
    tar -xf ./${ZULU_ARCH} -C ${INSTALL_DIR} && rm -f ${ZULU_ARCH} && \
    mv ${INSTALL_DIR}/${ZULU_DIR} ${JAVA_HOME} && \
    chmod a+x ${JAVA_HOME}/bin/* && \
    cd ${JAVA_HOME} && rm -rf src.zip demo sample man
```

emilm, 2019-05-20:
https://hub.docker.com/r/frolvlad/alpine-glibc this also have a very minimal base image. Java is big enough as it is, so everything that manages to slim it down is great

imrehg, 2019-05-20:
Am I mistaken, that the above `alpine-glibc` image is x86_64 only? As much as I can tell, the binaries are coming from https://github.com/sgerrand/docker-glibc-builder/releases and that only shows x86_64 versions. Thus the alpine glibc version would include quite a bit of work to expand across the other architectures we support as well. Just thinking out loud.

emilm, 2019-05-20:
Correct! it would be only available for a few architectures :/ in any case the 2nd RUN statement can be used with other distros than alpine.