balena-io-library/base-images #530: uname when cross-build from raspberry-pi-alpine gives armv7l instead of armv6l

Xavier-Yin, 2019-03-21:
## Summary
Using uname command during cross-building from raspberry-pi-alpine base image shows wrong architecture. (Same for raspberry-pi-alpine-python. I didn't try other language variants but I guess dealing with raspberry-pi-alpine may be enough)

## Steps to reproduce
build the following Dockerfile under macOS or Ubuntu18.04
```
FROM balenalib/raspberry-pi-alpine
RUN [ "cross-build-start" ]
RUN uname -m
RUN [ "cross-build-end" ]
```

## Expected results
"uname -m" gives armv6l architecture, just as same behavior when cross-building from raspberry-pi-debian base image

## Actual results
"armv7l"

nghiant2710, 2019-03-21:
@Xavier-Yin, `armv7l` is qemu default arm machine so without specifying `QEMU_CPU` environment variable, it will return that. `QEMU_CPU` in Debian base image is set to `arm1176` which will return `armv6l` as you already know, however it will break QEMU on Alpine Linux if that variable is set (`Illegal instruction`). That's why we don't set it in the Alpine Linux base images.

Xavier-Yin, 2019-03-21:
So what should we do if we plan to cross build from Alpine base images that is going to run on Raspberry Pi Zero (armv6), where we have several "compile from source" steps during docker build stage and the scripts involved "uname" for architecture?

nghiant2710, 2019-03-21:
@Xavier-Yin, I've raised your case with the team, we will discuss and get back to you as soon as we can.

Xavier-Yin, 2019-03-21:
@nghiant2710 Thank you!

nghiant2710, 2019-04-10:
@Xavier-Yin, we recently added `armv6 build slaves` to our builders, can you please try again?

closed 2019-05-24 by nghiant2710