balena-io-library/base-images #532: Debian Jessie pulled the plug on some package repositories

imrehg, 2019-03-26:
No more `jessie-updates`, etc, just `jessie`, see https://lists.debian.org/debian-devel-announce/2019/03/msg00006.html

This needs changes to our base images packag source list.

BraveUlysses81, 2019-03-29:
Yes. It's causing me an issue where I cannot build and push up to my repo. If I could just find where the pointers are to those calls, I could re-direct to the archive in which they are now located. It would at least be a temporary fix I could work with. As of now, I can't do anything.

git push balena
Counting objects: 741, done.
Delta compression using up to 8 threads.
Compressing objects: 100% (312/312), done.
Writing objects: 100% (741/741), 3.27 MiB | 1.48 MiB/s, done.
Total 741 (delta 414), reused 741 (delta 414)
remote: Resolving deltas: 100% (414/414), done.

[Info]     Starting build for lorawangw, user willie_witten
[Info]     Dashboard link: https://dashboard.balena-cloud.com/apps/1418133/devices
[Info]     Building on arm03
[Info]     Pulling previous images for caching purposes...
[Success]  Successfully pulled cache images
[main]     Step 1/14 : FROM resin/raspberrypi3-debian:latest AS buildstep
[main]      ---> 88ed09f4b4bf
[main]     Step 2/14 : WORKDIR /opt/ttn-gateway/
[main]      ---> 67c5a6c6397c
[main]     Removing intermediate container 95181ceb3999
[main]     Step 3/14 : RUN apt-get update &&     apt-get install wget build-essential libc6-dev git pkg-config protobuf-compiler libprotobuf-dev libprotoc-dev automake libtool autoconf python-dev python-rpi.gpio
[main]      ---> Running in 1d54fc0f4d7e
[main]     Get:1 http://archive.raspbian.org jessie InRelease [15.0 kB]
[main]     Get:2 http://archive.raspberrypi.org jessie InRelease [22.9 kB]
[main]     Get:3 http://security.debian.org jessie/updates InRelease [44.9 kB]
[main]     Ign http://deb.debian.org jessie InRelease
[main]     Ign http://deb.debian.org jessie-updates InRelease
[main]     Get:4 http://deb.debian.org jessie Release.gpg [2420 B]
[main]     Ign http://deb.debian.org jessie-updates Release.gpg
[main]     Get:5 http://deb.debian.org jessie Release [148 kB]
[main]     Ign http://deb.debian.org jessie-updates Release
[main]     Err http://deb.debian.org jessie-updates/main armhf Packages
[main]     Err http://deb.debian.org jessie-updates/main armhf Packages
[main]     Err http://deb.debian.org jessie-updates/main armhf Packages
[main]     Err http://deb.debian.org jessie-updates/main armhf Packages
[main]     Err http://deb.debian.org jessie-updates/main armhf Packages
[main]       404  Not Found
[main]     Get:6 http://archive.raspbian.org jessie/main armhf Packages [13.3 MB]
[main]     Get:7 http://archive.raspbian.org jessie/contrib armhf Packages [49.5 kB]
[main]     Get:8 http://archive.raspberrypi.org jessie/main armhf Packages [170 kB]
[main]     Get:9 http://archive.raspbian.org jessie/non-free armhf Packages [106 kB]
[main]     Get:10 http://archive.raspbian.org jessie/rpi armhf Packages [1297 B]
[main]     Get:11 http://archive.raspbian.org jessie/firmware armhf Packages [1208 B]
[main]     Get:12 http://deb.debian.org jessie/main armhf Packages [8898 kB]
[main]     Get:13 http://security.debian.org jessie/updates/main armhf Packages [794 kB]
[main]     Fetched 23.6 MB in 13s (1785 kB/s)
[main]     W: Failed to fetch http://deb.debian.org/debian/dists/jessie-updates/main/binary-armhf/Packages  404  Not Found
[main]     E: Some index files failed to download. They have been ignored, or old ones used instead.
[main]     
[main]     Removing intermediate container 1d54fc0f4d7e
[main]     The command '/bin/sh -c apt-get update &&     apt-get install wget build-essential libc6-dev git pkg-config protobuf-compiler libprotobuf-dev libprotoc-dev automake libtool autoconf python-dev python-rpi.gpio' returned a non-zero code: 100
[Info]     Uploading images
[Success]  Successfully uploaded images
[Error]    Some services failed to build:
[Error]      Service: main
[Error]        Error: The command '/bin/sh -c apt-get update &&     apt-get install wget build-essential libc6-dev git pkg-config protobuf-compiler libprotobuf-dev libprotoc-dev automake libtool autoconf python-dev python-rpi.gpio' returned a non-zero code: 100
[Error]    Not deploying release.

remote: error: hook declined to update refs/heads/master
To git.balena-cloud.com:willie_witten/lorawangw.git
 ! [remote rejected] master -> master (hook declined)
error: failed to push some refs to 'willie_witten@git.balena-cloud.com:willie_witten/lorawangw.git'

nghiant2710, 2019-03-29:
@BraveUlysses81 please change to the new `balenalib` base images since this issue has been fixed there or you can run this command `RUN sed -i '/jessie-updates/{s/^/#/}' /etc/apt/sources.list` before running `apt-get` as a workaround

fruxefarms, 2019-03-29:
I am using the new base images (I think) `balenalib/raspberrypi3-python:3.6-jessie-build` and encountering the same issue.

nghiant2710, 2019-03-29:
@fruxefarms we're building all base images to apply the fix for this issue so some images are not updated yet. I just checked the build servers and `balenalib/raspberrypi3-python:3.6-jessie-build` is not ready yet, will let you know when it's ready. Sorry for the inconvenience.

fruxefarms, 2019-03-29:
@nghiant2710 Thank you! I can confirm your suggested fix `RUN sed -i '/jessie-updates/{s/^/#/}' /etc/apt/sources.list` solves the issue for the interim.

nghiant2710, 2019-03-29:
@fruxefarms FYI the `balenalib/raspberrypi3-python:3.6-jessie-build` image is updated

fruxefarms, 2019-03-29:
@nghiant2710 Awesome! Thanks for the heads up!

nghiant2710, 2019-04-10:
fixed in https://github.com/balena-io-library/base-images/pull/533

closed 2019-04-10 by nghiant2710