balena-io-library/base-images #541: balenalib/armv7hf-debian:stretch Command not found

panadion, 2019-04-23:
Hello,
I try to build an image on top of balenalib/armv7hf-debian:stretch and there seems to be something wrong.

Tried to run with -it 
balenalib/armv7hf-debian:stretch-run
balenalib/armv7hf-debian:stretch-build
balenalib/armv7hf-debian:jessie 
and every image gets an error Command not found and gets exited.

nghiant2710, 2019-05-24:
@panadion, @dvldev can you please elaborate? Where did you run these images (on x86_64 or arm device)?

Since I can run them fine without any issue
```
docker run -ti --rm balenalib/armv7hf-debian:stretch bash
Unable to find image 'balenalib/armv7hf-debian:stretch' locally
stretch: Pulling from balenalib/armv7hf-debian
b0727b1ec48e: Already exists
1db5adee4b98: Already exists
527d42e6d470: Already exists
2767b8c7bcc6: Already exists
4781294745f0: Already exists
1394176b0103: Already exists
a29ca4eb50c9: Already exists
980b2f176d03: Already exists
Digest: sha256:69aeda61e270ff9f1672abb0ea7a989d03a457932493e168e9391dda170bb6d8
Status: Downloaded newer image for balenalib/armv7hf-debian:stretch
root@b13d6de487d6:/# uname -a
Linux b13d6de487d6 4.15.0-38-generic #41-Ubuntu SMP Wed Oct 10 10:59:38 UTC 2018 armv7l GNU/Linux
root@b13d6de487d6:/#
```

imrehg, 2019-07-09:
@panadion @dvldev do you still have his issue?

closed 2020-02-10 by nghiant2710