balena-io-library/base-images #547: Pip command not found in raspberrypi3-alpine-python images

penneydude, 2019-05-07:
We're seeing issues with pip not being found in the latest Python 2.7 images from Docker hub. 

To reproduce:

`docker run balenalib/raspberrypi3-alpine-python:2-edge-run pip`

The issue also exists for the `balenalib/raspberrypi3-alpine-python:2-latest` image

chrisys, 2019-05-10:
+1

The issue exists for `balenalib/%%BALENA_MACHINE_NAME%%-python:build` but not for `balenalib/%%BALENA_MACHINE_NAME%%-python:2.7.15-build`.

nghiant2710, 2019-05-13:
@penneydude, sorry for the inconvenience, we just had some version updates for the Python base images and there was a bug in the generation script that causes the issue. The fix was merged, deployed and we're rebuilding all the Python base images. Also, the images you mentioned are updated so you can use them now.

@chrisys most of the images are updated, so you shouldn't switch to the old image.

closed 2019-05-24 by nghiant2710