balena-io-library/base-images #563: Point `:latest` tag for Debian images to Buster

willswire, 2019-07-08:
https://www.debian.org/News/2019/20190706

imrehg, 2019-07-09:
@willswire please do not use the `:latest` tags in general, that's not stable, but use the appropriate other tags (including `buster` whenever it is possible)

What we **are** going to do, though, and it's in progress, that for Debian images if you don't specify a release, it will default to Buster (instead of the current Stretch). See docs (which we'll update when it happens) https://www.balena.io/docs/reference/base-images/base-images/#supported-architectures-distros-and-languages

(cc @nghiant2710 @shaunmulligan )

willswire, 2019-07-09:
@imrehg -- what I meant! Sorry, I guess my understanding of the `latest` tag was incorrect.

lurch, 2020-01-09:
This has been done now? ping @nghiant2710 and @garethtdavies

garethtdavies, 2020-01-09:
@lurch thanks yes, looks like this was done via this commit a while back: https://github.com/balena-io/contracts/commit/65e930f26be5afe52aac94b3b0911e13bf0c8512

closed 2020-01-09 by garethtdavies