balena-io-library/base-images #565: chromium-browser crashed in Docker environment.

alexzace, 2019-07-12:
Hi, below is my Dockerfile.
Its works in any Pi with any OS except Raspberry Pi 4 Buster. 

I figure out that once build the image with chromium-browser. 
The xserver cannot display the browser and xmessage. Its jus show black screen.
 
balenalib/raspberrypi3:buster also facing the same issue. 
Please assist on this. Thanks 

`FROM balenalib/raspberrypi3:stretch

RUN apt-get update && apt-get install -y \
        xserver-xorg-video-fbdev \
        xserver-xorg-input-all \
        openbox \
        x11-utils \
        chromium-browser \
        && rm -rf /var/lib/apt/lists/*
##  
##  add all required file
##
ENTRYPOINT ["/startup_x.sh"]
`

nghiant2710, 2019-07-12:
Hi @alexzace, so you're using a Raspberry Pi 4 right? Can you please try the aarch64 base images instead (`balenalib/raspberrypi3-64:stretch`) since the images above are ARMv7 so they might not be fully compatible.

alexzace, 2019-07-12:
ya i'm using Pi 4. Ok thanks let me try see.

alexzace, 2019-07-12:
after change the base image. 
failed to apt-get update 

`standard_init_linux.go:211: exec user process caused "exec format error"`

nghiant2710, 2019-07-12:
@alexzace, you need a 64bit OS image for it to work. Please select a 64bit OS image from balena dashboard.

alexzace, 2019-07-12:
Raspbian Buster OS does not have 64bit OS ?

nghiant2710, 2020-03-06:
yeah, I think they only have the `armv7` OS for now

closed 2022-06-22 by nghiant2710