balena-io-library/base-images #568: Add apt-get clean to install_packages script

chibby0ne, 2019-07-17:
For Ubuntu and Debian, apt is used as the installer behind the script `install_packages`:

https://github.com/balena-io-library/base-images/blob/75b59aa7266507ae51f8d67f7629f3c1e1e51ed8/balena-base-images/amd64/ubuntu/xenial/run/Dockerfile#L22-L45

The script runs apt-get with the option `--no-install-recommends` and deletes the `/var/lib/apt/lists/*` after running an `apt-get update` to avoiding bloating images with the additional packages and the list of packages from the servers.

But there are also the actual packages themselves (the cached packages) which are downloaded to: 
* `/var/cache/apt/archives/*`

Running an `apt-get clean` removes these directories. 
* `/var/cache/apt/archives/*`
* `/var/cache/apt/archives/partial/*`
* `/var/lib/apt/lists/partial/*`

Therefore I suggest we add this command to all the Ubuntu/Debian images' `install_package` script.

nghiant2710, 2019-07-23:
@chibby0ne, thank you very muck for the suggestion 👍 

I think it's safe to delete the whole `/var/cache/apt/archives` directory so we will update the `install_packages` script with this command `rm -r /var/lib/apt/lists /var/cache/apt/archives` to properly clean up everything.

chibby0ne, 2019-07-23:
@nghiant2710 I can submit a PR for this, I already starting working on this, but just wanted to validate with you guys.

I will also take a look for Alpine since I didn't really check that much there.

nghiant2710, 2020-03-06:
Closing as this issue is now resolved

closed 2020-03-06 by nghiant2710