balena-io-library/base-images #576: cross-build-start/end symlinks not available anymore in balenalib/rpi-raspbian

bbinet, 2019-08-13:
I cannot use cross-build scripts anymore with `balenalib/rpi-raspbian` images:

```
Sending build context to Docker daemon  8.192kB
Step 1/26 : FROM balenalib/rpi-raspbian:stretch
stretch: Pulling from balenalib/rpi-raspbian
Digest: sha256:3a548f0b566e89170618616468e660c2867a0dba6b2062e8161c620fde50a6ce
Status: Image is up to date for balenalib/rpi-raspbian:stretch
 ---> 3d40dbbb714c
Step 2/26 : RUN [ "cross-build-start" ]
 ---> Running in a77c26deaed3
OCI runtime create failed: container_linux.go:345: starting container process caused "exec: \"cross-build-start\": executable file not found in $PATH": unknown
Unable to find image 'helioslite/armv6hf-stretch-lsb_hindsight_build:1.0' locally
docker: Error response from daemon: manifest for helioslite/armv6hf-stretch-lsb_hindsight_build:1.0 not found: manifest unknown: manifest unknown.
```

Actually, `balenalib/rpi-raspbian` images does not seem to contain `cross-build-start`/`cross-build-end` symlinks anymore:

```
$ docker create --name="tmp_rpi_stretch" balenalib/rpi-raspbian:stretch
$ docker export tmp_rpi_stretch | tar t | grep bin | grep build
usr/bin/resin-xbuild
```

But it is still ok with image balenalib/armv7hf-debian:

```
$ docker create --name="tmp_armv7hf_stretch" balenalib/armv7hf-debian:stretch
$ docker export tmp_armv7hf_stretch | tar t | grep bin | grep build
usr/bin/cross-build-end
usr/bin/cross-build-start
usr/bin/resin-xbuild
```

nghiant2710, 2019-08-13:
hey @bbinet, we will add it back soon. Can you please add this line to your Dockerfile as a workaround for now?
```
RUN ln -s resin-xbuild /usr/bin/cross-build-start \
	&& ln -s resin-xbuild /usr/bin/cross-build-end
```

bbinet, 2019-08-13:
@nghiant2710 Thanks, it works when I manually add the symlinks.

closed 2020-03-06 by nghiant2710