balena-io-library/base-images #604: /usr/bin/cross-build-start: Invalid ELF image for this architecture

hailangvn, 2019-12-28:
Good day,
I encountered an issue and not sure what I did wrong. I am using Ubuntu 18.04, with balena-cli and docker.io installed.

```bash
EQUIPMENT_TYPE=raspberrypi3
EQUIP_ARCH=armv7hf
balena build --logs --deviceType ${EQUIPMENT_TYPE} --arch ${EQUIP_ARCH} --emulated
```

*docker-compose.yml
```
version: '2'

services:
  rpi3:
    container_name: rpi3
    build: ./rpi3
```

*rpi3/Dockerfile.template
```
FROM balenalib/%%BALENA_MACHINE_NAME%%-node:10.17-stretch-build
WORKDIR /usr/src/app
RUN [ "cross-build-start" ]
RUN [ "cross-build-end" ]
```

```
[Info]    Compose file detected
[Info]    Building for armv7hf/raspberrypi3
[Build]   Building services...
[Build]   rpi3 Preparing...
[Info]    Emulation is enabled
[Build]   rpi3 Step 1/5 : FROM balenalib/raspberrypi3-node:10.17-stretch-build
[Build]   rpi3  ---> ef6670066072
[Build]   rpi3 Step 2/5 : COPY [".balena/qemu-execve","/tmp/qemu-execve"]
[Build]   rpi3  ---> Using cache
[Build]   rpi3  ---> 72738c821580
[Build]   rpi3 Step 3/5 : WORKDIR /usr/src/app
[Build]   rpi3  ---> Using cache
[Build]   rpi3  ---> 696273fa068b
[Build]   rpi3 Step 4/5 : RUN ["/tmp/qemu-execve","-execve","/bin/sh","-c","cross-build-start"]
[Build]   rpi3  ---> Running in 69f163fff3ba
[Build]   rpi3 /usr/bin/cross-build-start: Invalid ELF image for this architecture
[Build]   Built 1 service in 0:16
[Error]   Build failed
The command '/tmp/qemu-execve -execve /bin/sh -c cross-build-start' returned a non-zero code: 255
                                                                                                  
Additional information may be available by setting a DEBUG=1 environment
variable: "set DEBUG=1" on a Windows command prompt, or "export DEBUG=1"
on Linux or macOS.
                                                                                                  
If you need help, don't hesitate in contacting our support forums at
https://forums.balena.io
                                                                                                  
For CLI bug reports or feature requests, have a look at the GitHub issues or
create a new one at: https://github.com/balena-io/balena-cli/issues/
```

hailangvn, 2019-12-28:
These are the versions. Please let me know if any other information needed.
```
docker --version
Docker version 18.09.7, build 2d0083d
balena --version
11.20.2
```

nghiant2710, 2019-12-28:
Hi @hailangvn,

As you already used `emulated` option, so you don't need `cross-build-start` and `cross-build-end` in your Dockerfile.

hailangvn, 2019-12-29:
Good day @nghiant2710,
Thanks a lot for the quick and precise instruction. Now I can build successfully.

closed 2019-12-29 by hailangvn