balena-io-library/base-images #606: `python:3` base image points to wrong version

lurch, 2020-01-09:
As discovered by @garethtdavies at https://github.com/balena-io/docs/pull/1285#discussion_r362953311

You'd _expect_ `balenalib/raspberrypi4-64-alpine-python:3-3.11-run` to be using the newest version of Python3 (i.e. 3.8.0) but for some reason it's using Python 3.6.9 :confused: 
I guessed at a _possible_ reason [here](https://github.com/balena-io/docs/pull/1285#discussion_r363715055)

lurch, 2020-01-17:
Looks like the version of Python included in a `python:3` base image is indeed semi-random :confused: 
`balenalib/raspberrypi4-64-alpine-python:3-3.11-run` now includes Python 3.7.5 (whereas https://github.com/balena-io/docs/pull/1322 says the latest version of Python 3.7.x is 3.7.6 and the latest version of Python 3.x is 3.8.1)

nghiant2710, 2020-01-17:
@lurch The new versions in the docs above are not available yet (still being built).

About the wrong version, yeah we do have a bug in `contrato` - the engine that produces base images combinations, each 3.x series will have a `3-*` tag and they will replace each other depend on which is the last tag built by Jenkins. I will fix it soon and in the meantime, I think it's better if you use the tag that has major and minor version e.g. `3.7-*` or full version then you will have an image with correct python version as described in the docs

lurch, 2020-01-17:
This isn't actually a feature I need myself, but when this bug is fixed I was going to suggest to @garethtdavies that it might be worth mentioning in the docs that `python` (without a version tag) will get the latest version of Python 2, but that `python:3` can be used to get the latest version of Python 3 (since Python 2 is now officially EOL).
:smiley: