balena-io-library/base-images #610: aarch64-debian-dotnet:3.1 not found

gerb0n, 2020-02-11:
Hi,

When browsing through the repo I can find the [dotnet 3.1 images](https://github.com/balena-io-library/base-images/tree/master/balena-base-images/dotnet/aarch64/debian/buster).

However when I'm using it inside a Dockerfile it says '**not found**' / '**manifest unknown**'.

What works: **FROM balenalib/aarch64-debian-dotnet:3.0-sdk**
What doesn't work: **FROM balenalib/aarch64-debian-dotnet:3.1-sdk**

It doesn't work for **runtime** or **aspnet** either.

nghiant2710, 2020-02-12:
@gerb0n, we're working on it, will let you know when the base images are ready

nghiant2710, 2020-02-18:
@gerb0n just want to let you know that the dotnet base image 3.1 are ready, please let us know if you need further support.

closed 2020-02-18 by nghiant2710