balena-io-library/base-images #612: Add custom `idle` command

tmigone, 2020-02-15:
A very common technique we use is to have a container idling. This could be either to troubleshoot an issue on an existing container or just as part of the development workflow when you don't want your container to exit but you are experimenting so don't have a proper `CMD` to keep it running.

The most common way of doing this is using this one-liner: 
- `CMD while : ; do echo "${MESSAGE=Idling...}"; sleep ${INTERVAL=600};`

See this project for more reference:
- https://github.com/balena-io-playground/balena-idling

It would be great to be able to achieve this with a custom command similar to [install_packages](https://github.com/balena-io-library/base-images/blob/master/balena-base-images/armv7hf/debian/stretch/run/Dockerfile#L26-L49). We could have for example `idle` or `balena_idle`, so that the one-liner turns into:
- `CMD balena_idle`

rahul-thakoor, 2020-02-17:
This would be very helpful. I've used this several times during development and debugging

nghiant2710, 2020-02-17:
Very interesting, @shaunmulligan what do you think? We can have like a Balena base images CLI and this command can be something like `balena idle`. We can add more commands to print image details or pull latest contract to check stack version and update...

tmigone, 2020-02-17:
That would be awesome! The one downside that I can think of is that having "another balena CLI" type tool could be confusing; even if the only thing that is the same is how you invoke the commands i.e: `balena push myApp` vs `balena idle`

shaunmulligan, 2020-02-25:
@nghiant2710 yeah this sounds like a good idea, although I agree with @tmigone that we shouldn't add yet another `balena` command as its weird enough with the `balena engine` commands. Perhaps bring it to the product call, I can't think of a good way to name it besides the `balena_idle` which doesn't allow us to expand the functionality much. I would like to use this as well, i think it would be pretty help for local mode projects where I could do something like `CMD main.py || balena_idle`

PR #616 (2020-04-15): Update base images to add onbuild messages, balena-info, balena-idle commands
Connects to https://github.com/balena-io/contracts/pull/150
Fixes #612
closed 2020-04-15 by nghiant2710