balena-io-library/base-images #627: Add coral sources to debian images

shaunmulligan, 2020-04-16:
We should add the coral sources to our coral base images (debian for sure, but maybe even ubuntu?) 

Basically for the default debian images we should have:
```
# Add Google Coral sources lists
RUN echo "deb https://packages.cloud.google.com/apt coral-edgetpu-stable main" | sudo tee /etc/apt/sources.list.d/coral-edgetpu.list
RUN curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -

# udev in the container to enable TPU correctly
ENV UDEV=1
COPY 99-tpu.rules /etc/udev/rules.d/99-tpu.rules
```

For the python images we should have:
```
# Install the TPU packages we will need
RUN install_packages libedgetpu1-std \
            libedgetpu-dev \
            python3-edgetpu \
            edgetpu-examples \
            python3-pip
```

the `99-tpu.rules` rules should contain:
```
"SUBSYSTEM=="usb",ATTRS{idVendor}=="1a6e",GROUP="plugdev" 
"SUBSYSTEM=="usb",ATTRS{idVendor}=="18d1",GROUP="plugdev"
```

nghiant2710, 2020-04-17:
hey @shaunmulligan, I think we should install all the packages marked as pre-installed here https://coral.ai/software/#debian-packages.

About `edgetpu-examples`, it contains the examples only so not sure if we should install it by default, maybe users should install it manually if needed

shaunmulligan, 2020-04-17:
@nghiant2710 yeah I agree!

nghiant2710, 2020-04-24:
Fixed in https://github.com/balena-io/contracts/commit/c621038eeb2cafd2fa68f5b3234ebd78b0062849

closed 2020-04-24 by nghiant2710