balena-io-library/base-images #631: balenalib/raspberrypi3-node run images not up to date

peterjuras, 2020-04-24:
It appears to me that for Node version 12.16(.2) there are only docker images for the build variant and not the run variant:

https://hub.docker.com/r/balenalib/raspberrypi3-node/tags?page=1&name=12.16.2

Any idea on why this might be? Is there in general an overview of how the images are named/tagged? I find it hard to find and correlate Dockerfiles from this repository to docker images in docker hub.

nghiant2710, 2020-04-24:
Hey @peterjuras,

You can find more details about our base images (how the images are named/tagged) in this docs https://www.balena.io/docs/reference/base-images/base-images/#balena-base-images.

About the missing tag, I will check our build servers and get back to you when v12.16.2 run image is available. Since we provide many images for the balena supported devices so a build can fail for no reason sometimes which results in a tag is missing.

nghiant2710, 2020-04-24:
@peterjuras, it should be ready now. Please let us know if you need further support.

closed 2020-04-24 by nghiant2710

peterjuras, 2020-04-24:
Awesome, thanks for the fast response!

peterjuras, 2020-05-16:
Hi,

it appears that the run image is missing again for the build of the newest node version of 12.16.3:

https://hub.docker.com/r/balenalib/raspberrypi3-node/tags?page=1&name=12.16.3

It is not urgent or necessary for me to update to the newest version, but I have the feeling that this happens everytime a new Node version comes out: The build versions are updated, the run versions however are not available.

mathroc, 2020-08-07:
I observe the same issue with the node-14 version, only -build version are available