balena-io-library/base-images #637: armv7hf-debian: /bin/sh shim breaks apt

AmedeeBulle, 2020-05-14:
The `/bin/sh` shim: 
https://github.com/balena-io-library/base-images/blob/47c88b4ad6d39ebe8bfc2fea02296e06c2a4e482/balena-base-images/armv7hf/debian/buster/build/Dockerfile#L138-L141
breaks `apt`

To reproduce:
```
docker run --rm -it balenalib/armv7hf-debian:buster-build /bin/bash
root@26979267a1d6:/# apt-get update
Get:1 http://deb.debian.org/debian buster InRelease [121 kB]
Get:2 http://deb.debian.org/debian buster-updates InRelease [49.3 kB]
Get:3 http://security.debian.org/debian-security buster/updates InRelease [65.4 kB]
0% [Working]   
```
It will _hang_ at this point -- actually `/bin/sh.real /bin/sh /usr/bin/apt-key  ....` will fork itself indefinitely (until it exhaust the process table)

As workaround I added 
```
RUN [[ -f /bin/sh.real ]] && rm /bin/sh && mv /bin/sh.real /bin/sh
```
as first step in my Dockerfiles...

Hexagenic, 2020-05-14:
We had a similar problem where autoconf of a library stopped working after upgrade of QEMU 4.0.0.0+balena to 4.0.0+balena2. Downgrading to `balenalib/raspberrypi3-debian:stretch-20200420` resolved the issue for our case.

nghiant2710, 2020-05-18:
@AmedeeBulle, @Hexagenic thank you for the report, we will investigate the issue and get back to you with details as soon as possible.

AmedeeBulle, 2020-05-18:
To clarify why/how it breaks: it happens when you don't enter the container with `/bin/sh`.

E.g.:
```
docker run --rm -it balenalib/armv7hf-debian:buster-build /bin/sh
```
won't give an issue as the shim kills itself on enter, while the following will cause the problem
```
docker run --rm -it balenalib/armv7hf-debian:buster-build /bin/bash
```

Similarly you get the same behavior with `Dockerfiles` when you don't use `/bin/sh` -- e.g. for this one, where I need `bash` as interpreter:

```
FROM balenalib/%%RESIN_MACHINE_NAME%%-debian:buster
SHELL ["/bin/bash", "-c"]
...
```

scotstan, 2020-08-04:
I was puzzled by this too. My Dockerfile was hanging on the first `apt-get update -q`. I  "fixed" it by triggering the shim through a non-bash built-in command:

`which which` 

Then the shim was gone, and my Docker build would continue. Smells hacky; there might be a better way to achieve this affect, namely `cat /.balena/messages/image-info` on the first start.

klutchell, 2021-12-21:
We've narrowed this down to an issue with the temporary shim we install over `/bin/sh` to print image details before restoring the original `sh` binary. `apt-get update` calls apt-key which execs another process (via the function ExecGPGV) which is the one that needs `sh` to be a shell. If `sh` is not an actual shell, thousands of processes will be spawned all trying to do the same thing (verify a key), and this could/does halt the system. My laptop crashed a few times while testing this if I forgot to kill the container process.

We probably need to re-evaluate the need for this temporary shim. In the meantime, some of the known workarounds are:

- call sh first so the binary is restored before apt calls it (`docker run --rm -it balenalib/raspberrypi:buster sh apt-get update`)
- don't specify bash as a shell at all, and allow the default shell to run once (`docker run --rm -it balenalib/raspberrypi:buster apt-get update`)
- start the container shell with sh and run apt after the shell starts (`docker run --rm -it balenalib/raspberrypi:buster sh)

Again this only impacts the very first execution of `/bin/sh`. If any other commands like an interactive shell run first, the apt command will end up using the real binary. It's only when skipping the initial sh shell that apt has to deal with it and fails.

PeterDaveHello, 2022-01-17:
I have the same problem even after downgrading to stretch-20200420 😅

klutchell, 2022-07-15:
Should be resolved by https://github.com/balena-io/contracts/pull/310