balena-io-library/base-images #641: python/raspberrypi4-64 Add Bluetooth Dependency

dereekb, 2020-05-27:
I've been trying to set up a Balena Python image for using bluetooth to scan BLE using the [python-bluezero project](https://github.com/ukBaz/python-bluezero).

The issue is that at runtime python cannot access the AF_BLUETOOTH socket because when python is compiled in these images it does not have the libbluetooth-dev installed (or more specifically, bluetooth.h and related files that become available at /usr/include/bluetooth after installing the dependency).

[This same issue](https://github.com/docker-library/python/pull/445) was documented in the official Docker-Library Python repo.

To test it you can add the following to a Dockerfile:

 `
RUN python3 -c "from socket import AF_BLUETOOTH"
`

If it was not compiled with libbluetooth-dev you will get:

`ImportError: cannot import name 'AF_BLUETOOTH' from 'socket' (/usr/local/lib/python3.7/socket.py)`

Without this it isn't possible to use the bluetooth module from Python. Simply calling install_packages on libbluetooth-dev in a docker file extending the base images doesn't work either, as it needs to be available at Python compile time.

So far I've tried adding the dependency to both the [build and run images](https://github.com/balena-io-library/base-images/tree/master/balena-base-images/python/raspberrypi4-64/debian/buster/3.7.7) for buster 3.7.7, but no luck so far. Someone who set up those images might know what I'm doing wrong.

Right now my work around is to work from the python:3.8-buster Docker image, but this isn't ideal when working with Balena.

nghiant2710, 2020-06-10:
Hey @dereekb, thank you for the report, we will update our Python build script and install necessary dependencies to support this. I will get back to you after the Python base images are rebuilt.

closed 2020-07-06 by nghiant2710

asherkhb, 2020-09-22:
I might be missing something, but I still am getting the error (cannot import name 'AF_BLUETOOTH' from 'socket' in  balenalib/raspberry-pi-python:3.7.7 - Have the images been rebuilt or is it still outstanding?