balena-io-library/base-images #644: NVIDIA/libglvnd fails to build in cross-build setup

shaunmulligan, 2020-06-12:
It appears that compiling `libglvnd` using the `cross-build` functionality in our containers fails with:
```
aclocal: error: echo failed with exit status: 1
```

However this builds on Docker4Mac correctly and also works on native aarch64 machine, so it seems something specific to the base images qemu emulation on Linux x86 machines.

Reproduction Dockerfile
```
FROM balenalib\/aarch64-ubuntu:xenial
RUN \[ "cross-build-start" \]

# I am just grabbing dependencies here.
RUN cd /tmp && apt-get update && apt-get install -y python-minimal git gettext-base gettext pkg-config libtool autotools-dev m4 automake autoconf libxext-dev libx11-dev x11proto-gl-dev

#
RUN cd /tmp && git clone https://github.com/NVIDIA/libglvnd
RUN cd /tmp/libglvnd && ./autogen.sh && ./configure && make && make install

RUN [ "cross-build-end" ]
```

roflcoopter, 2021-03-25:
I am also experiencing this, is there a fix?

AeroNotix, 2021-03-25:
Last time I ran into issues like this - we fixed it by using a specific version of qemu added to the container instead.

Since then, though, we have moved our build pipeline to using a native ARM builder with packet.net .

@roflcoopter if you reach out to support they will help you find the version of qemu to use.

roflcoopter, 2021-03-26:
> Last time I ran into issues like this - we fixed it by using a specific version of qemu added to the container instead.
>
> Since then, though, we have moved our build pipeline to using a native ARM [...]

Thanks for the reply! Reaching out to support, do i do that by creating a new issue you mean?

AeroNotix, 2021-03-26:
Yeah create a new ticket with balena. I can try to dig up the ticket I had with them when we saw this issue.