balena-io-library/base-images #645: balenalib/raspberrypi4-64-ubuntu:focal missing

marvinroger, 2020-06-25:
Hi there,

`balenalib/raspberrypi4-64-ubuntu:[focal,focal-run,focal-build]` is missing in the Docker Hub. I think that's a bug.

Thanks for your work!

dtischler, 2021-03-05:
@nghiant2710 Are you able to determine what might be wrong here?

dtischler, 2021-03-29:
Looks like these are now being properly generated:

The current links are:
https://hub.docker.com/layers/balenalib/raspberrypi4-64-ubuntu/focal/images/sha256-bc34c3530288f7d8cd509826958356ccaf528709b0980e5944ad1c47b9bdf747?context=explore

https://hub.docker.com/layers/balenalib/raspberrypi4-64-ubuntu/focal-run/images/sha256-bc34c3530288f7d8cd509826958356ccaf528709b0980e5944ad1c47b9bdf747?context=explore

https://hub.docker.com/layers/balenalib/raspberrypi4-64-ubuntu/focal-build/images/sha256-60a0bd80ce30400fc238a1e5b70e29dc4462c67fa2f9c53a4a10e73057b4a4bc?context=explore

Closing for now, thanks.

closed 2021-03-29 by dtischler